davidrapan/ha-solarman #596: All entities offline after sunset when the inverter switches off

corgan2222, 2025-05-17:
Hello,

First of all, many thanks for the great integration. It was really easy to set up and the values are all displayed. 
But unfortunately only until shortly after sunset. 
After that, the inverter seems to switch off and the integration shows all values as offline.
I think that the inverter switches itself off to save energy. But is this intentional? Is that how you all do it? The dashboards always look unattractive in the evening when the entities are offline.

Is there any Workaround for that problem? 
Can I prevent the shutdown of the inverter?
Do I have to update the inverter firmware?

Inverter: [SUN-M80G4-EU-Q0](https://www.deyeinverter.com/deyeinverter/2025/05/14/%E3%80%90b%E3%80%91datasheet_sun-m60-80-100g4-eu-q0_30240102201467_20250513_en.pdf) | 600-1000W | Single Phase | 2 MPPT | [Micro Inverter](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/deye_micro.yaml)

[image]

Thanks, Stefan

davidrapan, 2025-05-17:
Hi @corgan2222, yes, inverters w/o a battery usually turn off after sunset and there is nothing you can do about it and all sensors are unavailable at that time. It is designed this way simply because it is not possible to distinguish between a device disconnected during the day (for whatever reason) and a device disconnected during the night because it is turned off. Disconnect during the day would result in the device (sensors in HA) being stuck w/ the last known data for the entire duration of the disconnection, which could possibly skew your data.

The dashboard issue is annoying, I know, but it should be addressed in the dashboard. The HA frontend should have some mechanism for fallback values ​​(like zeros instead of warnings, etc.). So you can take this as an opportunity to propose a feature request (or show support for an existing one, if one already exists).

oxynatOr, 2025-05-17:
yea its pita!  so i dont use the entities provide by the addin for my dash. 
i have added temples for everything. an example:

```yaml
- name: PV Partyraum AC Output Power Losses
  unit_of_measurement: "W"
  device_class: power
  unique_id: "PV_Partyraum_AC_Output_Power_Losses"
  state_class: "measurement"
  icon: mdi:solar-panel-large
  state: >-
    {% set pl = states('sensor.partyraum_m2000_power_losses') %}
    {% if is_number(pl) %} 
    {{ float(pl,0)}}
    {% else %} 0.0
    {% endif %}
```

it will return the value when it is a number, otherwise it will display 0.0. 
with this no util_meter, temple... will freak out. 

you also can calc the Power ya self, there is no benefit tbh... 
```
- name: "PV Partyraum Panel1 Voltage"
  state: >
    {% set volt = states('sensor.partyraum_m2000_pv1_voltage') %}
    {% if is_number(volt)  %}
    {{ volt }}
    {% else %} 0.0
    {% endif %}
  unit_of_measurement: "V"
  device_class: voltage
  state_class: "measurement"
  unique_id: "PV_Partyraum_Panel1_Voltage"

- name: "PV Partyraum Panel1 Current"
  state: >
    {% set amp = states('sensor.partyraum_m2000_pv1_current') %}
    {% if is_number(amp) %}
    {{ amp }}
    {% else %} 0.0
    {% endif %}
  unit_of_measurement: "A"
  device_class: "current"
  state_class: "measurement"
  unique_id: "PV_Partyraum_Panel1_Current"

- name: PV Partyraum Panel1 AC Output Power
  unit_of_measurement: "W"
  device_class: power
  unique_id: "PV_Partyraum_Panel1_AC_Output_Power"
  state_class: "measurement"
  icon: mdi:solar-panel-large
  state: >-
    {% set volt = states('sensor.pv_partyraum_Panel1_Voltage') %}
    {% set amp = states('sensor.pv_partyraum_Panel1_Current') %}
    {% if is_number(volt) and is_number(amp) %} 
    {{ float(volt,0) * float(amp,0) }}
    {% else %} 0.0
    {% endif %}
```

you can do funny thing with it, like an average. 
```
- name: PV Average AC Output Power Losses
  unit_of_measurement: "W"
  device_class: power
  unique_id: "pv_average_ac_output_power_losses"
  state_class: "measurement"
  icon: mdi:solar-power
  state: >-
    {% set wk1 = states('sensor.pv_werkstatt1_ac_output_power_losses') %}
    {% set pr = states('sensor.pv_partyraum_ac_output_power_losses') %}
    {% set pwk2 = states('sensor.pv_werkstatt2_ac_output_power_losses') %}
    {% if is_number(wk1) and is_number(pr) and is_number(pwk2) %}
    {{ (float(wk1,0) + float(pr,0) + float(pwk2,0))/3 }}
    {% else %} 0.0
    {% endif %}
```

hope this will help a bit.

corgan2222, 2025-05-17:
Thank you for your answers. I was afraid that it would need such workarounds. Thanks @oxynatOr for the code. I can customize the rest.

oxynatOr, 2025-05-18:
> You may also find that you get logging every evening and morning as the inverter goes off and online especially in the winter where the transition from night to day is very slow. I've added a couple of log filters to remove these.

yea, thats why you need to as a Template/"Filter" to each sensor. 
because of the German LAW and the rules of the commune, it was less work for me to install 4 "Balkonkraftwerke" and have a mini PV( ~4kwhp)

i'm open to share all sensors. just let me know.

Eskander, 2025-05-29:
> The dashboards always look unattractive in the evening when the entities are offline.

The second option, which I went with, is to simply hide the entities from the dashboard when they're unavailable:

```yaml
type: tile
entity: sensor.inverter_l1_voltage
name: Voltage
icon: mdi:sine-wave
color: amber

visibility:
  - condition: state
    entity: sensor.inverter_l1_voltage
    state_not: unavailable
```

The condition could be tied to Sun state instead:

```yaml
visibility:
  - condition: state
    entity: sun.sun
    state: above_horizon
```