davidrapan/ha-solarman #604: Deye SG05 configuration

Kuciak96, 2025-05-22:
### Is your feature request related to a problem?

Hi, I'm using this integration for Deye sg04 and it's perfectly fine. I wanted to use also for new PV installation for my brother using SG05 LV inverter but it shows values completely incorrectly (like millions of kWh of production for new PV plant).

### Describe the solution you'd like

Can you tell me if you plan to prepare configuration file also for new series of inverters? It looks like some differences in data flow. Thank you for your project!

### Describe alternatives you've considered

If your update will take time then I will switch to esphome and I will get data through RS port.

davidrapan, 2025-05-22:
Hi @Kuciak96,

> If your update will take time then I will switch to esphome and I will get data through RS port.

If this device requires new profile you will face the same issue.

Can  you provide the debug log?

Kuciak96, 2025-05-22:
> Hi [@Kuciak96](https://github.com/Kuciak96),
> 
> > If your update will take time then I will switch to esphome and I will get data through RS port.
> 
> If this device requires new profile you will face the same issue.
> 
> Can you provide the debug log?

In case of ESPhome I will write it on my own based on project for sg04.

I will upload here debug log later, I need to connect remotely to my brother.

davidrapan, 2025-05-22:
> In case of ESPhome I will write it on my own based on project for sg04.

If is SG05 different from SG04 (as you are implying) then you will still need modbus register list for this device, it is the same as w/ this integration. Communication goes always through modbus at the end.

Kuciak96, 2025-05-22:
> > In case of ESPhome I will write it on my own based on project for sg04.
> 
> If is SG05 different from SG04 (as you are implying) then you will still need modbus register list for this device, it is the same as w/ this integration. Communication goes always through modbus in the end.

YOu're right and I'm aware of it. But I've assumed that I will find modbus description like for sg04. Now I know it is not available yet.

robertsze, 2025-05-27:
I have a SUN-20K-SG05LP3-EU-SM2 and it works completely fine with the deye_p3 configuration, just for your information

Kuciak96, 2025-05-27:
I've got an update. I've completely removed integration and installed it again (did it already 3 times...) and now it works fine. I have no idea what's the reason, before no any faults called etc. now it works fine.
My inverter is 15k LV 3 phase.

closed 2025-05-27 by davidrapan