davidrapan/ha-solarman #606: SmartLoad Grid On stopped working after 25.05.20 update

linean, 2025-05-22:
### Description

I've updated integration from [v25.05.12](https://github.com/davidrapan/ha-solarman/releases/tag/v25.05.12) to [v25.05.20](https://github.com/davidrapan/ha-solarman/releases/tag/v25.05.20) and `SmartLoad Grid On` stopped to work. I'm using `deye_p3` config. By default switch is disabled even IO mode is set to SmartLoad. When I comment this section:

```
   enabled_lookup:
        io_mode_select: [Generator, SmartLoad]
```
I can toggle the switch but it fails with

```
Failed to perform the action switch/turn_on. unsupported operand type(s) for &: 'NoneType' and 'int'
````

### Reproduction Steps

1. Update to v25.05.12
2. Set IO mode to SmartLoad
3. Toggle SmartLoad Grid On switch

### Log

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 250, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/solarman/switch.py", line 70, in async_turn_on
    value = self._to_native_value(self._value_on)
  File "/config/custom_components/solarman/switch.py", line 55, in _to_native_value
    return (self._attr_native_value & ~(1 << self._value_bit)) | (value << self._value_bit)
            ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for &: 'NoneType' and 'int'
```

### Version

25.05.12

### Home Assistant Core Version

2025.5.2

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-22:
Hi @linean, thank you for your report, this is most likely fixed alongside [fix: Conditional use of generator port in calculation of losses](https://github.com/davidrapan/ha-solarman/commit/9554ef33bca9302809fdce85632c8c0b5c51ed62)

Reading `enabled_lookup` was broken before the fix and the unsupported operand error proly means it never read the value from the inverter because it is one of many switches that are under one register and are toggled using bit masking.

linean, 2025-05-22:
> the unsupported operand error proly means it never read the value from the inverter because it is one of many switches that are under one register and are toggled using bit masking.

Does it mean it was not loaded yet but should be loaded at some point? I'm trying to debug `_attr_native_value` and it seems to never become available.

davidrapan, 2025-05-22:
Did you restart HA?

linean, 2025-05-22:
I did, many times. Other features from integration seems to work fine. I only have troubles with this switch

linean, 2025-05-22:
I was playing with Deye app and out of the sudden switch started to work. Maybe it was some glitch on my end. Thanks for pointing me to the fix 🤝  I'm gonna close this issue for now and revert to .12.

closed 2025-05-22 by linean