davidrapan/ha-solarman #608: After update to v25.05.20 DEYE entities become Unavailable very often

iwannatalk, 2025-05-23:
### Description

After update to v25.05.20 DEYE entities become Unavailable very often, seems like the issue is because of attached errors in logs

### Reproduction Steps

Update to v25.05.20

### Log

Error fetching Inverter data: Invalidate complete dataset - 14260 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}
Error fetching Inverter data: Invalidate complete dataset - 12900 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}
Error fetching Inverter data: Invalidate complete dataset - 13970 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}
Error fetching Inverter data: Invalidate complete dataset - 13620 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}
Error fetching Inverter data: Invalidate complete dataset - 13040 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}

ParameterParser.try_parse: data: {(3, 0): [6, 1, 260, 12852, 12342, 12848, 14385, 13620, 0, 0, 8, 7688, 0, 11038, 12290, 4230, 0, 8193, 49205, 0, 53392, 3, 515, 5, 1, 0, 0, 59, 1, 10, 1, 258, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 65535, 6405, 5899, 7728, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 96, 1, 100, 100, 65535, 0, 1000, 1, 0, 65535, 1, 65535, 1, 1, 65535, 65535, 65535, 1, 0, 65535, 0, 65535, 65535, 65535, 0, 1, 5760, 5760, 5520, 200, 1600, 2, 90, 0, 5, 40, 40, 0, 1, 0, 25, 990, 8, 10, 15, 1600, 1800, 1700, 240, 0, 4900, 30], (3, 125): [20, 4900, 10, 30, 0, 1, 5500, 0, 0, 5100, 95, 5400, 100, 1, 5, 3, 0, 1, 2500, 65535, 0, 255, 0, 0, 700, 1100, 1300, 1900, 2300, 2500, 2500, 2499, 2500, 2500, 2500, 2900, 2900, 2900, 2900, 2900, 2900, 15, 15, 20, 20, 20, 20, 1, 1, 1, 1, 1, 1, 11962, 65535, 60, 0, 0, 0, 1, 2650, 1850, 5150, 4800, 0, 1500, 1200, 100, 1000, 0, 0, 40000, 0, 20000, 0, 800, 0, 35, 3, 460, 0, 55, 3, 41486, 0, 4608, 0, 92, 100, 61, 4311, 65287, 1260, 0, 0, 0, 0, 0, 0, 0, 4100, 100, 12289, 3819, 5, 0, 2500, 65535, 5500, 65535, 0, 0, 1], (3, 340): [2750], (3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 12573, 41265, 0, 0, 0, 0, 171, 192, 53548, 0, 47355, 0, 1, 0, 18585, 1, 21, 0, 334, 7022, 2, 317, 0, 0, 0, 0, 59559, 0, 0, 0, 0, 0, 1250, 1512, 0, 0, 1, 0, 0, 0, 0, 0, 2048, 1, 7, 0, 0, 0, 0, 0, 0], (3, 586): [1260, 4329, 61, 0, 64453, 63033, 200, 0, 65535, 0, 64536, 0, 2407, 2406, 2400, 65517, 4, 0, 1, 2, 9, 12, 0, 5004, 109, 149, 103, 24, 22, 5, 1, 1, 1, 3, 0, 34, 1, 2, 9, 12, 0, 2402, 2397, 2406, 150, 440, 390, 328, 1036, 891, 2255, 2255, 5004, 1000, 329, 1038, 900, 2267, 2387, 2399, 2388, 0, 0, 0, 329, 1038, 900, 2267, 2267, 5004, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 411, 912, 0, 0, 3345, 122, 5927, 153, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], (3, 10032): [13104, 12340, 12337, 12336, 13891, 14386, 12336, 12343, 2, 0, 1, 0, 1, 1, 157, 4311, 29756, 72, 60, 100, 3377, 8, 9, 3365, 1, 3, 64, 66, 1, 5, 63, 8]}, definition: {'name': 'PV Power', 'alt': 'DC Power', 'mppt': 1, 'description': 'Combined power of all inputs', 'class': 'power', 'state_class': 'measurement', 'uom': 'W', 'rule': 1, 'digits': 0, 'scale': 10, 'sensors': [{'registers': [672], 'code': 3, 'scale': 10}, {'registers': [673], 'code': 3, 'scale': 10}, {'registers': [674], 'code': 3, 'scale': 10}, {'registers': [675], 'code': 3, 'scale': 10}], 'validation': {'min': 0, 'max': 12800, 'invalidate_all': None}, 'icon': 'mdi:solar-power-variant', 'platform': 'sensor', 'key': 'pv_power_sensor', 'registers': [672, 673, 674, 675], 'group': 'PV', 'update_interval': 5, 'code': 3} [ValueError: Invalidate complete dataset - 13230 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}]
ParameterParser.try_parse: data: {(3, 0): [6, 1, 260, 12852, 12342, 12848, 14385, 13620, 0, 0, 8, 7688, 0, 11038, 12290, 4230, 0, 8193, 49205, 0, 53392, 3, 515, 5, 1, 0, 0, 59, 1, 10, 1, 258, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 65535, 6405, 5899, 7728, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 96, 1, 100, 100, 65535, 0, 1000, 1, 0, 65535, 1, 65535, 1, 1, 65535, 65535, 65535, 1, 0, 65535, 0, 65535, 65535, 65535, 0, 1, 5760, 5760, 5520, 200, 1600, 2, 90, 0, 5, 40, 40, 0, 1, 0, 25, 990, 8, 10, 15, 1600, 1800, 1700, 240, 0, 4900, 30], (3, 125): [20, 4900, 10, 30, 0, 1, 5500, 0, 0, 5100, 95, 5400, 100, 1, 5, 3, 0, 1, 2500, 65535, 0, 255, 0, 0, 700, 1100, 1300, 1900, 2300, 2500, 2500, 2499, 2500, 2500, 2500, 2900, 2900, 2900, 2900, 2900, 2900, 15, 15, 20, 20, 20, 20, 1, 1, 1, 1, 1, 1, 11962, 65535, 60, 0, 0, 0, 1, 2650, 1850, 5150, 4800, 0, 1500, 1200, 100, 1000, 0, 0, 40000, 0, 20000, 0, 800, 0, 35, 3, 460, 0, 55, 3, 41486, 0, 4608, 0, 92, 100, 61, 4311, 65291, 1260, 0, 0, 0, 0, 0, 0, 0, 4100, 100, 12289, 3819, 5, 0, 2500, 65535, 5500, 65535, 0, 0, 1], (3, 340): [2750], (3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 12573, 41265, 0, 0, 0, 0, 171, 192, 53548, 0, 47355, 0, 1, 0, 18585, 1, 21, 0, 334, 7022, 2, 317, 0, 0, 0, 0, 59559, 0, 0, 0, 0, 0, 1250, 1512, 0, 0, 1, 0, 0, 0, 0, 0, 2048, 1, 7, 0, 0, 0, 0, 0, 0], (3, 586): [1260, 4329, 61, 0, 64456, 63040, 200, 0, 65535, 0, 64536, 0, 2403, 2403, 2398, 65517, 5, 0, 0, 0, 10, 10, 0, 5000, 108, 149, 104, 24, 21, 5, 1, 1, 1, 3, 0, 32, 0, 0, 10, 10, 0, 2399, 2394, 2407, 160, 450, 380, 327, 1035, 890, 2252, 2252, 5000, 64536, 327, 1035, 900, 2262, 2384, 2396, 2394, 0, 0, 0, 327, 1035, 900, 2262, 2262, 5000, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 411, 913, 0, 0, 3396, 122, 5976, 153, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], (3, 10032): [13104, 12340, 12337, 12336, 13891, 14386, 12336, 12343, 2, 0, 1, 0, 1, 1, 157, 4311, 29749, 72, 60, 100, 3377, 8, 9, 3365, 1, 3, 64, 66, 1, 5, 63, 8]}, definition: {'name': 'PV Power', 'alt': 'DC Power', 'mppt': 1, 'description': 'Combined power of all inputs', 'class': 'power', 'state_class': 'measurement', 'uom': 'W', 'rule': 1, 'digits': 0, 'scale': 10, 'sensors': [{'registers': [672], 'code': 3, 'scale': 10}, {'registers': [673], 'code': 3, 'scale': 10}, {'registers': [674], 'code': 3, 'scale': 10}, {'registers': [675], 'code': 3, 'scale': 10}], 'validation': {'min': 0, 'max': 12800, 'invalidate_all': None}, 'icon': 'mdi:solar-power-variant', 'platform': 'sensor', 'key': 'pv_power_sensor', 'registers': [672, 673, 674, 675], 'group': 'PV', 'update_interval': 5, 'code': 3} [ValueError: Invalidate complete dataset - 13240 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}]
ParameterParser.try_parse: data: {(3, 0): [6, 1, 260, 12852, 12342, 12848, 14385, 13620, 0, 0, 8, 7688, 0, 11038, 12290, 4230, 0, 8193, 49205, 0, 53392, 3, 515, 5, 1, 0, 0, 59, 1, 10, 1, 258, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 65535, 6405, 5899, 7954, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 96, 1, 100, 100, 65535, 0, 1000, 1, 0, 65535, 1, 65535, 1, 1, 65535, 65535, 65535, 1, 0, 65535, 0, 65535, 65535, 65535, 0, 1, 5760, 5760, 5520, 200, 1600, 2, 90, 0, 5, 40, 40, 0, 1, 0, 25, 990, 8, 10, 15, 1600, 1800, 1700, 240, 0, 4900, 30], (3, 125): [20, 4900, 10, 30, 0, 1, 5500, 0, 0, 5100, 95, 5400, 100, 1, 5, 3, 0, 1, 2500, 65535, 0, 255, 0, 0, 700, 1100, 1300, 1900, 2300, 2500, 2500, 2499, 2500, 2500, 2500, 2900, 2900, 2900, 2900, 2900, 2900, 15, 15, 20, 20, 20, 20, 1, 1, 1, 1, 1, 1, 11962, 65535, 60, 0, 0, 0, 1, 2650, 1850, 5150, 4800, 0, 1500, 1200, 100, 1000, 0, 0, 40000, 0, 20000, 0, 800, 0, 35, 3, 460, 0, 55, 3, 41486, 0, 4608, 0, 92, 100, 61, 4311, 65283, 1260, 0, 0, 0, 0, 0, 0, 0, 4100, 100, 12289, 3819, 5, 0, 2500, 65535, 5500, 65535, 0, 0, 1], (3, 340): [2750], (3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 12573, 41265, 0, 0, 0, 0, 171, 192, 53548, 0, 47355, 0, 1, 0, 18585, 1, 21, 0, 334, 7022, 2, 317, 0, 0, 0, 0, 59559, 0, 0, 0, 0, 0, 1250, 1512, 0, 0, 1, 0, 0, 0, 0, 0, 2048, 1, 7, 0, 0, 0, 0, 0, 0], (3, 586): [1260, 4329, 61, 0, 64451, 63029, 200, 0, 65535, 0, 64536, 0, 2406, 2408, 2400, 65517, 4, 0, 0, 0, 10, 10, 0, 5000, 108, 149, 104, 24, 21, 5, 1, 1, 1, 3, 0, 32, 0, 0, 10, 10, 0, 2399, 2394, 2407, 160, 450, 380, 327, 1035, 890, 2252, 2252, 5000, 64536, 327, 1035, 900, 2262, 2384, 2396, 2394, 0, 0, 0, 327, 1035, 900, 2262, 2262, 5000, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 411, 913, 0, 0, 3396, 122, 5976, 153, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,  [... cut ...]  'scale': 10}], 'validation': {'min': 0, 'max': 12800, 'invalidate_all': None}, 'icon': 'mdi:solar-power-variant', 'platform': 'sensor', 'key': 'pv_power_sensor', 'registers': [672, 673, 674, 675], 'group': 'PV', 'update_interval': 5, 'code': 3} [ValueError: Invalidate complete dataset - 13250 of pv_power_sensor validation failed: {'min': 0, 'max': 12800, 'invalidate_all': None}]

### Version

v25.05.20

### Home Assistant Core Version

2025.5.2

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-23:
Can someone provide debug log from latest version?

davidrapan, 2025-05-23:
Hi @iwannatalk, thank you for your report, here is a [fix: PV Power validation for Deye 3P HV](https://github.com/davidrapan/ha-solarman/commit/1b47b6f0e9fe183ea839f40a73d79f50987dcc6a)