davidrapan/ha-solarman #611: LSW3 Wifi stick logger keeps on disconnecting

dmagyar, 2025-05-23:
I'm getting fed up with my wifi stick logger, I've updated firmware as well so it became somewhat better but not 100%. Symptom is that it disconnects from wifi and it stays that way for a few minutes. It eventually comes back. Wifi is 100% at the spot I even brought a dedicated AP next to the inverter.

My question is - can I maybe use this excellent integration with an alternative modbus-tcp device connected to the inverter? I'm thinking esphome on an esp32 with an rs485-ttl converter.

davidrapan, 2025-05-24:
Hi @dmagyar, can you show me the debug log then?

But yes, this integration does work w/ standard Modbus TCP. There is transport protocol select in device configuration.

dar3khudy, 2025-05-25:
I'm getting similar https://github.com/davidrapan/ha-solarman/issues/586 but with LSE3 (so no WiFi but ethernet).

dar3khudy, 2025-05-26:
In my case it is 50s, it is working better than with 5s but it is still not stable, what is weird network connection is just fine without HA, but with HA there is always ping missing after connection lose with solarman - no idea, I manually removed almost all data from reading, except power and energy and seams to be a bit better, but why it is still unknow.

kkmaslowski, 2025-05-27:
@davidrapan @dar3khudy @dmagyar For me latest version (v25.05.26) is fully stable with 5s interval. No more disconnections! Good job David!

[image]

PV-Joe, 2025-05-29:
100% stable with 5 seconds here.
Never had an issue with any version.

kkmaslowski, 2025-06-02:
It started happening again on latest version v25.06.02. It was rock stable on v25.05.26. 

[image]

davidrapan, 2025-06-03:
Well, there was a bug w/ a side effect of "hiding" disconnects, but it should have only purely cosmetic effect in your case as all sensors should remain available.

kkmaslowski, 2025-07-10:
> Well, there was a bug with the side effect of "hiding" the disconnections, but in your case it should only have a purely cosmetic effect, as all sensors should remain available. Simply put, what's happening behind the scenes hasn't changed, it's just now more visible through the connection sensor.

For me it looked much better when disconnections were hidden

closed 2025-08-25 by github-actions[bot]