davidrapan/ha-solarman #612: Solarman performance degraded with version v25.05.20 (tried with v25.05.22)

AndyNew2, 2025-05-25:
### Description

Performance degraded with latest version.
I had installed version v25.05.12, which could update my data at around 5sec update rate. 
It sometimes had been slower to 7 - 20 seconds, however more than 90% of the updates it was around 5 seconds.

With the latest version this performance degraded, and I had best 7 seconds (never reaching the desired 5 seconds anymore) and mainly around 10 seconds, very often update rate about 12 - 16 seconds.

I used my custom sofar definition derived from sofarg3d

### Reproduction Steps

upgrade solarman integration from v25.05.12 to v25.05.22.
Having an Sofar Hybrid Inverter
Used custom sofar definition with multiple rates to optimize update rates depending on needs.
See attached file

### Log

[zcs_azzurro-hyd3ph_5K_20K.yaml.txt](https://github.com/user-attachments/files/20430348/zcs_azzurro-hyd3ph_5K_20K.yaml.txt)

### Version

25.05.22

### Home Assistant Core Version

2025.05.3

### Home Assistant Operating System Version

14.2

davidrapan, 2025-05-25:
Hi @AndyNew2, can you please provide a debug log? This is inverter's profile.

AndyNew2, 2025-05-26:
Here a log file with the old version v25.05.20 for comparance. I will upload later on a log with the newest version and will report...

[home-assistant_solarman_2025-05-26T04-19-50.744Z.log](https://github.com/user-attachments/files/20434789/home-assistant_solarman_2025-05-26T04-19-50.744Z.log)

AndyNew2, 2025-05-26:
See now the log for the newest version v25.05.24. It seemed to improve slightly compared to v25.05.22. It toggels now around 7seconds to 11-12seconds. But still away from the 5 seconds, I am used in v25.05.12...

[home-assistant_solarman_2025-05-26T04-29-53.519Z.log](https://github.com/user-attachments/files/20434845/home-assistant_solarman_2025-05-26T04-29-53.519Z.log)

davidrapan, 2025-05-26:
Try new version. 😉

davidrapan, 2025-05-26:
Great! 🎉