davidrapan/ha-solarman #624: Ich kann den TM-L8M nicht einfügen

Robocat1712, 2025-05-27:
### Is your feature request related to a problem?

Moin Leute. 

Ich habe von Astro-Energy den Wechselrichter TM-L8M. Leider kann ich den mit egal was ich versuche nicht in Home Assistant binden. Mit der App löuft es problemlos. Was muss ich da genau auswählen? Bei mir wird nicht ein einziger Wert angezeigt.

### Describe the solution you'd like

Moin Leute. 

Ich habe von Astro-Energy den Wechselrichter TM-L8M. Leider kann ich den mit egal was ich versuche nicht in Home Assistant binden. Mit der App läuft es problemlos. Was muss ich da genau auswählen? Bei mir wird nicht ein einziger Wert angezeigt.

### Describe alternatives you've considered

Habe schon alles mögliche versucht

stokifan, 2025-05-28:
Hi,
i use a TM-L800Mi from Astro-Energy. I tried for a while using pysolarmanv5 to read it. It didn't work the way it was advertised there, however when scanning the addresses I found that for certain ones he returns an unexpected long frame in the solarman v5 format. pysolarmanv5 does not process it, even though I found that it is correctly formatted.

b'\xa5\x8b\x00\x10\x15^\xf3<serial>\x02\x01fY\n\x003\x1e\x00\x00\x00\xfb,h\x01\x04x\x01$\x01$\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00\x00\t\x14\x00\x00\x13\x88\x00\x00\x00\xd1\x00\x00\x00\x00\x00\x00\x00\x00\x00>\x00\x00\x01\xd0\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xa9\xb8\x84\x15'

I read the string and I found there are many non-null values in there. Comparing them to the values reported in the Solarman App, I found which is which. I processed the data with a python script below, which uses some code from pysolarmanv5 for checksum calculation, and some other code I found online for the MTU-Bus CRC16 calculation. I assume that might work for yours as well. You need to fill in the ip-address and the serial number of your device.

```
#!/usr/bin/python
import socket
import requests
import sys
import time

def _calculate_checksum(data: bytes) -> int:
        """
        Calculate checksum on all bytes

        :param data: data for checksum calculation
        :type data: bytes
        :return: Checksum value of all bytes
        :rtype: int

        """
        checksum = 0
        for d in data:
            checksum += d & 0xFF
        return int(checksum & 0xFF)

def _calculate_v5_frame_checksum(frame: bytes) -> int:
        """
        Calculate checksum on all frame bytes except head, end and checksum

        :param frame: V5 frame for checksum calculation
        :type frame: bytes
        :return: Checksum value of V5 frame
        :rtype: int

        """
        return _calculate_checksum(frame[1:-2])

def init_table( ):
    # Initialize the CRC-16 table,
    #   build a 256-entry list, then convert to read-only tuple
    global table

    lst = []
    for i in range(256):
        data = i << 1
        crc = 0
        for j in range(8, 0, -1):
            data >>= 1
            if (data ^ crc) & 0x0001:
                crc = (crc >> 1) ^ 0xA001
            else:
                crc >>= 1

        lst.append( crc)

    table = tuple( lst)
    return

def calcCRC16( st, crc):
    """Given a bunary string and starting CRC, Calc a final CRC-16 """
    for ch in st:
        crc = (crc >> 8) ^ table[(crc ^ ch) & 0xFF]
    return crc

init_table()
host = <ip-address>
port = 8899
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.settimeout(5)
s.connect((host, port))
frame=0
PV1_V=0
PV2_V=0
PV1_A=0
PV2_A=0
PDC_W=0
AC_A=0
AC_V=0
FRQ_HZ=0
TMP_C=0
MPP=0
PG_W=0
E_KWH=0
serial=<serial-number>
sdata=bytearray(b'\xa5\x17\x00\x10\x45\x5e\x00'+(serial & 0xFF).to_bytes(1)+((serial >> 8) & 0xFF).to_bytes(1)+((serial >> 16) & 0xFF).to_bytes(1)+(serial >> 24).to_bytes(1)+b'\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x04'+frame.to_bytes(1)+b'\x00\x00\x78\xf0\x36\x54\x15')
crc16=calcCRC16(sdata[26:32],0xFFFF)
sdata[32]=crc16 & 0xFF
sdata[33]=(crc16 >> 8) & 0xFF
sdata[-2]=_calculate_v5_frame_checksum(sdata)
rep=1
for avg in range (0,rep):
  s.sendall(sdata)
  data = s.recv(1024)
  print('Received', repr(data))
  PV1_V+=0.1*(data[25+3]*256+data[25+4])
  PV2_V+=0.1*(data[25+5]*256+data[25+6])
  PV1_A+=0.03*(data[25+13]*256+data[25+14])
  PV2_A+=0.03*(data[25+15]*256+data[25+16])
  AC_A+=0.01*(data[25+21]*256+data[25+22])
  AC_V+=0.1*(data[25+25]*256+data[25+26])
  FRQ_HZ+=0.01*(data[25+29]*256+data[25+30])
  TMP_C+=0.1*(data[25+47]*256+data[25+48])-26.3
  MPP+=data[25+49]*256+data[25+50]
  PG_W+=0.1*(data[25+33]*256+data[25+34])
  E_KWH+=0.990/57*(data[25+43]*256+data[25+44])
#  time.sleep(10)
PDC_W=(PV1_V*PV1_A+PV2_V*PV2_A)/rep/rep
resstr='PDC_W={0:1.2f}'.format(PDC_W)
resstr=resstr+',PV1_V={0:1.2f}'.format(PV1_V/rep)
resstr=resstr+',PV2_V={0:1.2f}'.format(PV2_V/rep)
resstr=resstr+',PV1_A={0:1.2f}'.format(PV1_A/rep)
resstr=resstr+',PV2_A={0:1.2f}'.format(PV2_A/rep)
resstr=resstr+',AC_A={0:1.2f}'.format(AC_A/rep)
resstr=resstr+',AC_V={0:1.2f}'.format(AC_V/rep)
resstr=resstr+',FRQ_HZ={0:1.2f}'.format(FRQ_HZ/rep)
resstr=resstr+',TMP_C={0:1.2f}'.format(TMP_C/rep)
resstr=resstr+',MPP={0:1d}'.format(MPP//rep)+'i'
resstr=resstr+',PG_W={0:1.2f}'.format(PG_W/rep)
resstr=resstr+',E_KWH={0:1.2f}'.format(E_KWH/rep)
print(resstr)
s.close()

```

stokifan, 2025-05-28:
Hi david, no, it was something with unpack and not enough bytes reserved or so. I'm not so deep in Python to figure it out.

stokifan, 2025-05-29:
Hi david, I just tried, you are right, when asking for 60 registers it works!

stokifan, 2025-05-29:
MPP refers probably to the number MPPTs used (two panels for instance). The WG_W is probably the generated Power. I'm still watching this, there also the register before that seems to be related to Power, maybe one is apparent and the other effective Power or so.

davidrapan, 2025-05-31:
It would be the first device with such hacky encoding, so I really don't think it's right.

davidrapan, 2025-06-01:
Do you run HA?

davidrapan, 2025-06-03:
The disconnect between manual V*A and "power" as a standalone is pretty much "normal" across all devices, so I wouldn't think too much about it if it's a few watts here and there. 😉

stokifan, 2025-06-03:
Yes, Solarman app does a lot of averaging, most noticible on the delays of temperature and DC-Current, but that power data is delayed too. The app says it sends data every 5 min but measures only one minute (likely the minute before sending). There is fluctuation on the data, so everyone sampling differently gets slightly modified results. As long as it is around 20-50W accurate that is fine for me.

emmericp, 2025-06-06:
Here's a full day of data: https://docs.google.com/spreadsheets/d/1iJmwLz3k5GR_AbsJva6d1XgWPZnbDb8qmww5QmklqmI/edit?usp=sharing

It's missing data for about an hour as packet capturing failed. It sometimes uploads more often than expected, in most cases there is a 1:1 match between data points in the app and the capture, but sometimes it uploads twice within 2-3 seconds and the app only reports one data point. I removed these since I can't be sure about the exact mapping.

Anyhow, I haven't done detailed analysis yet, but a quick look:

# Temperature

[image]

[image]

```
Linear Fit: temperature = 0.112 * raw + -31.856
Polynomial Degree 2 Coefficients: [ 6.37531550e-05  3.70116005e-02 -1.07351111e+01]
Polynomial Degree 3 Coefficients: [ 1.91278133e-07 -2.60355966e-04  2.16371220e-01 -4.32025366e+01]
Polynomial Degree 4 Coefficients: [ 2.34855902e-10 -3.43817238e-07  1.90923971e-04  4.94589623e-02
 -2.03634560e+01]
Polynomial Degree 5 Coefficients: [ 5.52023219e-13 -1.33278001e-09  1.41899993e-06 -7.90118237e-04
  3.19644642e-01 -4.98225603e+01]
Exponential Fit: temperature = 46.843 * exp(0.001 * raw) + -61.025
```

Clearly missing something. 4 or 5 degrees can really fit anything, so that's not the actual answer. The pattern of the residuals at around 500 looks sus, maybe it's not a continuous function as suspected above. 

Just from looking at the data the relationship 1 = 0.1 °K looks correct at lower temperatures, but obviously wrong at higher temperatures.

To be continued.

# Current

A bit more messy. Looks like we need both current and voltage together to make it work.

There are multiple cases of the device reporting the exact same current across different packets, but the app showing different currents (and power) for these entries. But voltage and current together seem to be able to predict current and power shown in the app. Search for raw current value 120 in the data to see examples.

To be continued.

stokifan, 2025-06-07:
Hi,
I played with this data. Current comes very close to a square function, so I will try that for being simple. When I sorted, I noted some inconsistency, for same bin data sometimes there are multiple values. I tuned my simple math for this temperature data, but that may need to change once higher temp data is available. My setup has pretty much the same power limit, so I will not be able to extend the range here, unfortunately.

[solar 2025-06-06.xlsx](https://github.com/user-attachments/files/20640091/solar.2025-06-06.xlsx)

emmericp, 2025-06-07:
> When I sorted, I noted some inconsistency, for same bin data sometimes there are multiple values.

Once you add the voltage this goes away; same raw voltage and raw current always translates to the same current.

stokifan, 2025-06-10:
Hi, I tried just squaring the current with a constant multiplied, and it worked reasonably well, but that voltage impact was still a bit annoying. So I added a voltage correction by subtracting the reported voltage from a fixed voltage, applied a constant factor, and subtracted this from the current value. Constants were changed to match:

I = 0.000652*I_solarman^2*(1-(29-V_solarman)*0.042)

This gives very satisfactorily matching results and using just simple math. See the yellow and blue lines below, SM_PDC is from Solarman, PDC is my direct read. The fluctuations are caused from the independent sampling between Solarman and me, and there is a delay for the yellow one due to uploading data and downloading. I also updated the spreadsheet linked above.

For the other available graph in the data (PG in red), I plotted here in comparison to AC_VA, which is just AC voltage * AC current. You can see that it is basically the same, with PG having a bit better resolution, therefore not fully identical. So it is the apparent power. There seems to be like 20W of blind power with very low light, then it changes into real power as levels rise up. That is my interpretation, unfortunately cos phy is not reported directly.

[image]

Marcel2221, 2025-06-28:
How are you reading the values – via Modbus? And which port are you using?
My inverter is now set up via Solarman and regularly sends data there. I can also ping it through its internal IP address, but I can’t establish any Modbus connection.

stokifan, 2025-06-28:
See my latest python-script attached. Fill in your ip-address and your serial-number, and it should show you the values:

pi@raspberrypi:~ $ ./TM-L800Mi.py
Dur:0.17693638801574707

astro-e_low_data,URL=<ip-address> PDC_W=112.33,AC_VA=57.80,PV1_V=29.00,PV2_V=29.00,PV1_A=1.90,PV2_A=1.97,AC_A=0.25,AC_V=231.20,FRQ_HZ=50.00,TMP_C=27.41,MPP=2i,PG_W=56.80,E_KWH=0.12
Tot_dur:0.1871812343597412

[TM-L800Mi.zip](https://github.com/user-attachments/files/20959286/TM-L800Mi.zip)

ralphcastro504, 2025-07-04:
Good day Guys, 

Thanks a lot for all you guys collaboration here. 

I see that the Experimental support is added to the inverter definition for The hoe assistant integration. 

I tried to configure mine to check the values but I could not, following message appears,  

[image]

TheWerewolf1984, 2025-07-31:
Ich habe das gleich eProblem, hier was ich im Logfile gefunden habe:

`Einrichtungsfehler, wird erneut versucht: The server has accepted the requests and it processing it, but a long duration of time will be required to do so.`

dalori, 2025-09-06:
Same here. I can get a response with the Python script mentioned above, but the integration seems to timeout, even though the logs show results as above. Happy to help in debugging.

tomayac, 2025-12-16:
I have the [TM-L800M](https://www.astro-e.com.cn/?lm1/87.html) and got it to work with this custom config file (placed it in `/homeassistant/custom_components/solarman/inverter_definitions/tm-l800m.yaml`) based on [`astro-energy_micro.yaml`](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/astro-energy_micro.yaml). It reports data, but it's somewhat off, likely because of not doing yet what https://github.com/davidrapan/ha-solarman/issues/624#issuecomment-2952278581 suggests. I had Gemini 3 Pro massage it until it spat out data feeding it the debug logs. Maybe this helps someone come up with a final config that works more accurately.  

```yaml
info:
  manufacturer: Astro-Energy
  model: TM-L series (Hybrid)

default:
  update_interval: 60
  max_size: 60
  code: 0x04
  min_span: 60 # Keeps the connection stable

requests_fine_control:
  requests:
    - code: 0x04
      start: 0x00
      end: 0x3B
      count: 0x3C

parameters:
  # --- PV SIDE (DC) ---
  - group: PV
    items:
      - name: PV Power
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 0.1
        rule: 1
        registers: [0x0000]
        icon: "mdi:solar-power-variant"
        # This calculates Total DC Power (PV1 + PV2) automatically
        sensors:
          - registers: [0x0000] # PV1 Volts
            multiply:
              registers: [0x0005] # PV1 Amps
              scale: 0.01
          - registers: [0x0001] # PV2 Volts
            multiply:
              registers: [0x0006] # PV2 Amps
              scale: 0.01

      - name: "PV1 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x0000]

      - name: "PV1 Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.01
        rule: 1
        registers: [0x0005]

      - name: "PV2 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x0001]

      - name: "PV2 Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.01
        rule: 1
        registers: [0x0006]

  # --- GRID SIDE (AC) ---
  - group: Grid
    items:
      - name: "Grid Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.01
        rule: 1
        registers: [0x0009]

      - name: "Grid Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x000B]

      - name: "Grid Frequency"
        class: "frequency"
        state_class: "measurement"
        uom: "Hz"
        scale: 0.01
        rule: 1
        registers: [0x000D]

      - name: "AC Output Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 0.1
        rule: 1
        registers: [0x000F]
        icon: 'mdi:solar-power'

      - name: "Internal Temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 0.1
        rule: 1
        registers: [0x0016]

      - name: "Total Energy"
        class: "energy"
        state_class: "total_increasing"
        uom: "kWh"
        scale: 0.01
        rule: 1
        registers: [0x0014]

  # --- INFO ---
  - group: Info
    items:
      - name: "MPPT Count"
        rule: 1
        registers: [0x0017] # Should show "2"
      
      # Dummy sensor to force the full read
      - name: "Block_End_Marker"
        class: ""
        scale: 1
        rule: 1
        registers: [0x003B]
        is_str: true
```

[image]

[image]