davidrapan/ha-solarman #632: After restarting the wifi, the integration is not working until i "reload" it

vladmihociu, 2025-05-30:
### Description

Hello.

I have an automation that each night at 03:00 i restart all the Wifi Meshes from the house. 
Until the last release, the integration had no issues. It worked even after the wifi restart :) but with the latest version (25.05.26), I have to Reload the integration to make it work again :). Happened 3 times in the last 3 days (exactly after the wifi restart).

The internal IP is not being changed. It remaiuns the same.

[image]

### Reproduction Steps

.

### Log

[image]

[image]

### Version

25.05.26

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-30:
Hi @vladmihociu, I am unable to reproduce the issue. Can you please provide a debug log?

Dompii, 2025-05-31:
> ### Description
> Hello.
>
> I have an automation that each night at 03:00 i restart all the Wifi Meshes from the house. Until the last release, the integration had no issues. It worked even after the [...]

I have same issue :(

Dompii, 2025-06-01:
I collected the logs...
Hopefully i do it in right way 😅😅

[Solarman.txt](https://github.com/user-attachments/files/20538437/Solarman.txt)

vladmihociu, 2025-06-01:
first time doing it.
hope it is usefull

[debuginfo_vlad.txt](https://github.com/user-attachments/files/20538478/debuginfo_vlad.txt)

davidrapan, 2025-06-01:
Thanks, here is [fix: Retry decorator catching CancelledError](https://github.com/davidrapan/ha-solarman/commit/bc9842edb92aa6b3a14be984829f8779280d6c72)

closed 2025-06-01 by davidrapan