davidrapan/ha-solarman #635: Deye "Load L1 Power" value is off by 1 decimal point

redfishheaven, 2025-05-31:
### Description

Load L1 Power in Home Assistant appear to show the correct value as Deye Cloud reports. but the value is off by one decimal point

As in the screenshot. Deye Cloud report 850W of power, but in HA it shows 85W. And that offset is constant for the entire data log.

[image]

### Reproduction Steps

My Inverter model is SUN-10K-SG05LP1-EU (3 String Version)
On the Device info it show SG0*LP1

The inverter has one PYLON type battery connected.
Load is connected to UPS port from the inverter.

Configuration of the integration is in this picture.
[image]

### Log

I've only setup the integration as above. Not modified anything yet.
All the other values from the integration seems to read correctly. reports same as Deye Cloud.

### Version

25.05.26

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

redfishheaven, 2025-05-31:
> Hi [@redfishheaven](https://github.com/redfishheaven), switch on the scale modifier. 😉

Can you guide me how to do that?

davidrapan, 2025-05-31:
You can see it on the image you uploaded to the issue, under Additional options.

redfishheaven, 2025-05-31:
Value is reading correctly now. Thanks for fast response. And sorry for my bad not seeing that option. 🥹

[image]

davidrapan, 2025-05-31:
No worries. 😉

closed 2025-05-31 by davidrapan