davidrapan/ha-solarman #641: Ethernet logger on Sofarsolar ME3000SP broken

jdeluyck, 2025-06-03:
### Description

When updating to 25.06.02 my integration now complains "The data address received in the request is not an allowable address for the server. "

I've ready switched it from "TCP" to "Modbus TCP" but nothing changed.

### Reproduction Steps

Upgraded and restarted HA

### Log

```
2025-06-03 13:22:00.253 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-22T14:00:33.381594+00:00', 'data': {'serial': 2104761096}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JFPQAWB5ASFF794WEPZ0TG31', 'minor_version': 9, 'modified_at': '2025-06-03T11:13:08.786043+00:00', 'options': {'host': '192.168.38.10', 'lookup_file': 'sofar_wifikit.yaml', 'transport': 'modbus_tcp'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'SofarSolar ME3000-SP', 'unique_id': None, 'version': 1})
2025-06-03 13:22:00.253 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-06-03 13:22:00.253 DEBUG (MainThread) [custom_components.solarman.coordinator] _async_setup
2025-06-03 13:22:00.253 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 192.168.38.10
2025-06-03 13:22:00.254 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.38.10
2025-06-03 13:22:00.256 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: ('192.168.38.10', 48899) [192.168.38.10, 34EAE7D9EC04, 2104761096]
2025-06-03 13:22:00.256 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: sofar_wifikit.yaml
2025-06-03 13:22:00.256 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-06-03 13:22:00.369 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 3, min_span: 25, max_size: 125, digits: 6, attributes: {'mod': 0, 'mppt': 4, 'l': 3, 'pack': -1}
2025-06-03 13:22:00.372 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Scheduling 4 query requests: [{'code': 3, 'start': 512, 'end': 597, 'count': 86}, {'code': 3, 'start': 4608, 'end': 4623, 'count': 16}, {'code': 4, 'start': 4272, 'end': 4284, 'count': 13}, {'code': 4, 'start': 8192, 'end': 8203, 'count': 12}] #0
2025-06-03 13:22:00.372 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Request 03 ❘ 0x03 ~ 0512 ❘ 0x0200: {'count': 86}
2025-06-03 13:22:00.372 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: fa 05 00 00 00 06 01 03 02 00 00 56
2025-06-03 13:22:00.374 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Successful connection!
2025-06-03 13:22:01.475 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: fa 05 00 00 00 03 01 83 02
2025-06-03 13:22:01.475 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 8b 42 00 00 00 06 01 03 02 00 00 56
2025-06-03 13:22:02.708 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 8b 42 00 00 00 03 01 83 02
2025-06-03 13:22:02.708 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Request 03 ❘ 0x03 ~ 0512 ❘ 0x0200: {'count': 86}
2025-06-03 13:22:02.708 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 9c 78 00 00 00 06 01 03 02 00 00 56
2025-06-03 13:22:04.043 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 9c 78 00 00 00 03 01 83 02
2025-06-03 13:22:04.043 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 0b 3d 00 00 00 06 01 03 02 00 00 56
2025-06-03 13:22:05.174 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 0b 3d 00 00 00 03 01 83 02
2025-06-03 13:22:05.174 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Closing connection
2025-06-03 13:22:05.174 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SofarSolar ME3000-SP data in 4.803 seconds (success: False)
2025-06-03 13:22:05.174 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-06-03 13:22:05.174 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Closing connection
```

### Version

25.06.02

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

davidrapan, 2025-06-03:
Hi @jdeluyck, can you try 2 things for me?
1) Add `max_size: 20` into `sofar_wifikit`'s default section
2) Remove whole `group: Other` from `sofar_wifikit`'s parameters section

_Do the first, if nothing changes, try the second._

jdeluyck, 2025-06-03:
@davidrapan  Neither makes a difference - I've restarted HA each time.

davidrapan, 2025-06-03:
Can you show me the debug log for both? (Just wanna make sure the changes did anything)

jdeluyck, 2025-06-03:
with `max_size: 20`

```
2025-06-03 14:25:31.274 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-22T14:00:33.381594+00:00', 'data': {'serial': 2104761096}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': 'xxx', 'minor_version': 9, 'modified_at': '2025-06-03T11:13:08.786043+00:00', 'options': {'host': '192.168.38.10', 'lookup_file': 'sofar_wifikit.yaml', 'transport': 'modbus_tcp'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'SofarSolar ME3000-SP', 'unique_id': None, 'version': 1})
2025-06-03 14:25:31.274 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-06-03 14:25:31.274 DEBUG (MainThread) [custom_components.solarman.coordinator] _async_setup
2025-06-03 14:25:31.274 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 192.168.38.10
2025-06-03 14:25:31.275 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.38.10
2025-06-03 14:25:31.277 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: ('192.168.38.10', 48899) [192.168.38.10, 34EAE7D9EC04, 2104761096]
2025-06-03 14:25:31.277 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: sofar_wifikit.yaml
2025-06-03 14:25:31.277 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-06-03 14:25:31.376 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 3, min_span: 25, max_size: 20, digits: 6, attributes: {'mod': 0, 'mppt': 4, 'l': 3, 'pack': -1}
2025-06-03 14:25:31.378 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Scheduling 8 query requests: [{'code': 3, 'start': 512, 'end': 531, 'count': 20}, {'code': 3, 'start': 532, 'end': 551, 'count': 20}, {'code': 3, 'start': 552, 'end': 571, 'count': 20}, {'code': 3, 'start': 572, 'end': 584, 'count': 13}, {'code': 3, 'start': 592, 'end': 597, 'count': 6}, {'code': 3, 'start': 4608, 'end': 4623, 'count': 16}, {'code': 4, 'start': 4272, 'end': 4284, 'count': 13}, {'code': 4, 'start': 8192, 'end': 8203, 'count': 12}] #0
2025-06-03 14:25:31.378 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Request 03 ❘ 0x03 ~ 0512 ❘ 0x0200: {'count': 20}
2025-06-03 14:25:31.378 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: bd 74 00 00 00 06 01 03 02 00 00 14
2025-06-03 14:25:31.381 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Successful connection!
2025-06-03 14:25:32.493 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: bd 74 00 00 00 03 01 83 02
2025-06-03 14:25:32.493 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 01 28 00 00 00 06 01 03 02 00 00 14
2025-06-03 14:25:33.623 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 01 28 00 00 00 03 01 83 02
2025-06-03 14:25:33.623 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Request 03 ❘ 0x03 ~ 0512 ❘ 0x0200: {'count': 20}
2025-06-03 14:25:33.623 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 39 d3 00 00 00 06 01 03 02 00 00 14
2025-06-03 14:25:34.753 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 39 d3 00 00 00 03 01 83 02
2025-06-03 14:25:34.754 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 56 76 00 00 00 06 01 03 02 00 00 14
2025-06-03 14:25:35.884 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 56 76 00 00 00 03 01 83 02
2025-06-03 14:25:35.884 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Closing connection
2025-06-03 14:25:35.885 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SofarSolar ME3000-SP data in 4.507 seconds (success: False)
2025-06-03 14:25:35.885 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-06-03 14:25:35.885 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Closing connection
```

jdeluyck, 2025-06-03:
Without the `group: Other` section

```
2025-06-03 14:29:53.574 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-22T14:00:33.381594+00:00', 'data': {'serial': 2104761096}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': 'xxx', 'minor_version': 9, 'modified_at': '2025-06-03T11:13:08.786043+00:00', 'options': {'host': '192.168.38.10', 'lookup_file': 'sofar_wifikit.yaml', 'transport': 'modbus_tcp'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'SofarSolar ME3000-SP', 'unique_id': None, 'version': 1})
2025-06-03 14:29:53.574 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-06-03 14:29:53.574 DEBUG (MainThread) [custom_components.solarman.coordinator] _async_setup
2025-06-03 14:29:53.574 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 192.168.38.10
2025-06-03 14:29:53.574 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.38.10
2025-06-03 14:29:53.576 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: ('192.168.38.10', 48899) [192.168.38.10, 34EAE7D9EC04, 2104761096]
2025-06-03 14:29:53.576 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: sofar_wifikit.yaml
2025-06-03 14:29:53.576 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-06-03 14:29:53.648 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 3, min_span: 25, max_size: 125, digits: 6, attributes: {'mod': 0, 'mppt': 4, 'l': 3, 'pack': -1}
2025-06-03 14:29:53.650 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Scheduling 4 query requests: [{'code': 3, 'start': 512, 'end': 597, 'count': 86}, {'code': 3, 'start': 4608, 'end': 4623, 'count': 16}, {'code': 4, 'start': 4272, 'end': 4284, 'count': 13}, {'code': 4, 'start': 8192, 'end': 8203, 'count': 12}] #0
2025-06-03 14:29:53.650 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Request 03 ❘ 0x03 ~ 0512 ❘ 0x0200: {'count': 86}
2025-06-03 14:29:53.650 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 3f c0 00 00 00 06 01 03 02 00 00 56
2025-06-03 14:29:53.652 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Successful connection!
2025-06-03 14:29:54.778 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 3f c0 00 00 00 03 01 83 02
2025-06-03 14:29:54.778 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 85 25 00 00 00 06 01 03 02 00 00 56
2025-06-03 14:29:55.909 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 85 25 00 00 00 03 01 83 02
2025-06-03 14:29:55.909 DEBUG (MainThread) [custom_components.solarman.device] [192.168.38.10] Request 03 ❘ 0x03 ~ 0512 ❘ 0x0200: {'count': 86}
2025-06-03 14:29:55.909 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 04 54 00 00 00 06 01 03 02 00 00 56
2025-06-03 14:29:57.039 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 04 54 00 00 00 03 01 83 02
2025-06-03 14:29:57.040 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] SENT: 53 51 00 00 00 06 01 03 02 00 00 56
2025-06-03 14:29:58.271 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] RECV: 53 51 00 00 00 03 01 83 02
2025-06-03 14:29:58.272 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Closing connection
2025-06-03 14:29:58.272 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SofarSolar ME3000-SP data in 4.623 seconds (success: False)
2025-06-03 14:29:58.273 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-06-03 14:29:58.273 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.38.10] Closing connection
```

davidrapan, 2025-06-03:
And you updated from version?

jdeluyck, 2025-06-03:
Since no version seemed to work I went down to verify that my ethernet stick was still connected correctly, and saw that the COM light was out. After reseating it it came back, and it works now fine. With the latest version.

I do have it on Modbus TCP instead of normal TCP - perhaps the original update confused it?

davidrapan, 2025-06-03:
Phew, great, because this "problem" confused the hell out of me.

But it's interesting how it manifested, anyway TCP means that the Solarman Modbus RTU encapsulation will be used, but with ethernet loggers there is an automatic switch to standard Modbus TCP.