davidrapan/ha-solarman #643: ZCS Azzurro HYD 3/6k ZSS HP loses entities with Sofar G3 profile

gmcinalli, 2025-06-05:
### Description

Hi, I have a **ZCS Azzurro HYD 3/6k ZSS HP** with 4 batteries, before I used the inverter definition `zcs_azzurro-hyd-zss-hp.yaml` and it worked perfectly.

With the release of [v25.06.02](https://github.com/davidrapan/ha-solarman/releases/tag/v25.06.02), I switched to `sofar_g3.yaml` selecting battery packs 1 as stated [here](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices) but some entities disappeared, for example the section with _"Device Serial Number"_ or _"Device Fault"_ are no longer shown.

Looking in the yaml files the registry values ​​of the previous file and current one match, why the entities do not appear anymore?

Before I had about 68 entities, now only 50.

### Reproduction Steps

Update to [v25.06.02](https://github.com/davidrapan/ha-solarman/releases/tag/v25.06.02) and use `sofar_g3` profile with ZCS Azzurro HYD 3/6k ZSS HP.

### Log

[home-assistant_solarman_2025-06-05T08-09-03.331Z.log](https://github.com/user-attachments/files/20605662/home-assistant_solarman_2025-06-05T08-09-03.331Z.log)

### Version

v25.06.02

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

davidrapan, 2025-06-05:
Hi @gmcinalli, for example Device Serial Number is now an attribute attached to Device State sensor. Are you sure that Device Fault is missing?

gmcinalli, 2025-06-05:
@davidrapan `Battery EPS Buffer`, `Battery EOD`, `Battery DOD` are some examples of missing entities that before were present.

```yaml
      - name: "Battery DOD"
        update_interval: 30
        state_class: "measurement"
        uom: "%"
        rule: 1
        registers: [0x104D]
        icon: "mdi:battery"

      - name: "Battery EOD"
        update_interval: 30
        state_class: "measurement"
        uom: "%"
        rule: 1
        registers: [0x104E]
        icon: "mdi:battery"

      - name: "Battery EPS Buffer"
        update_interval: 30
        state_class: "measurement"
        uom: "%"
        rule: 1
        registers: [0x1052]
        icon: "mdi:battery-low"
```

gmcinalli, 2025-06-05:
Ok, thank you, waiting for the next release so!

davidrapan, 2025-06-05:
Maybe you still had some of those Fault X sensors from before their merge?

davidrapan, 2025-06-05:
> Another thing I did in the additional settings was set the number of phases to 1, because mine is a single phase inverter, so the other phases do not appear.

👍 Maybe that's it.

I have a question. When you did the update. Was it able to load the device without changing the configuration? There is a "redirect" that should handle this, so I'm curious if it worked?

gmcinalli, 2025-06-05:
> I have a question. When you did the update. Was it able to load the device without changing the configuration? There is a "redirect" that should handle this, so I'm curious if it worked?

No, I saw the redirect in the commit, but I don't think it worked, I had some old sensors that had become "unavailable" and then seeing this I proceeded to reconfigure the extension from scratch.

But before doing so I clicked "Configure" and I saw that there was nothing selected as a profile, so I thought I had to reconfigure from scratch.

davidrapan, 2025-06-05:
Oh yeah, it's definitely better if you do that, but it can be loaded without any changes and the redirect is also able to set default parameters. Proly the changed entities were unavailable. 😉

gmcinalli, 2025-06-05:
Ok, so to recap, there will be a new release that will re-add the entities that I don't see anymore, as mentioned above, Battery EOD etc...

davidrapan, 2025-06-05:
If you don't want to wait, you can easily replace the file contents in your HA with this "newer" version. 😉

davidrapan, 2025-06-05:
No, here it's just for battery packs = 0: w/o batteries, battery packs = 1: w/ batteries.

gmcinalli, 2025-06-05:
Yes, but now they're incorporated into "Device fault", right? But still there are not 68.

davidrapan, 2025-06-05:
It does: 6x sensors which are now attributes + 6x **Fault** sensors + 2x **L2 *** and 2x **L3 *** sensors. 😉

gmcinalli, 2025-06-05:
You are a dragon! 🐉