davidrapan/ha-solarman #647: Error setuping Inverter: CRCError('CRC validation failed.')

mdpf73, 2025-06-06:
### Description

after updating to version 25.06.06 no inverter is detected even by manually selecting all types of inverters from the list, my zcs azzurro inverter is not detected, before the update everything worked very well!!!

### Reproduction Steps

Configurazione non riuscita, riproverà: Error setuping Inverter: CRCError('CRC validation failed.')

### Log

Configurazione non riuscita, riproverà: Error setuping Inverter: CRCError('CRC validation failed.')

### Version

25.06.06

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

davidrapan, 2025-06-06:
Hi @mdpf73, can you please enable debug logging and attach the file? Thank you.

kakupl, 2025-06-06:
@davidrapan  @mdpf73  i changed value 0x0214 to registers: [0x0485]
and problem is resolved 

      - name: "Power losses"
        description: Includes consumption of the inverter device itself as well AC/DC conversion losses
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 1
        digits: 0
        scale: 10
        uint: enforce
        sensors:
          - signed:
            registers: [0x0606]
          - registers: [0x05C4]
          - operator: subtract
            signed:
            registers: [0x0485]

davidrapan, 2025-06-06:
So it was the same as #648