davidrapan/ha-solarman #648: Error connecting to Azzurro inverter with new sofar_g3.yaml

lcavalli, 2025-06-06:
### Description

After updating integration it stopped connecting to the inverter. Tried also to reconfigure the device and also to add new device from scratch.

[image]

### Reproduction Steps

Update to new version of the integration. No further steps are required.

### Log

The error message is: "The data address received in the request is not an allowable address for the server."

### Version

25.06.06

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

lakatosmarlonbrando, 2025-06-06:
The same as @lcavalli on SofarSolar 10kW HYD 3PH. Replaced sofar_g3hyd.yaml with the one from the previous release, now it works.

sarmbruster, 2025-06-06:
Having here a Sofar HYD 4000 EP which used to be configured as zcs_azzurro-hyd-zss-hp.yaml in previous version. Getting the same error message as above. Switching config to sofar_g3.yaml does not help and still shows the very same message.

Additionally I do have a Sofar 3300 TL which continues to work configured as sofar_lsw3.yaml

sarmbruster, 2025-06-06:
As a workaround I've picked the old  zcs_azzuro-hyd-hp.yaml (https://github.com/davidrapan/ha-solarman/blob/v25.05.26/custom_components/solarman/inverter_definitions/zcs_azzurro-hyd-zss-hp.yaml) and copied it into   `/config/custom_components/solarman/inverter_definitions/custom/`. In the UI I've selected this profile instead of `sofar_g3.yaml`. With this I do get all the measurements back.

Swarozyc, 2025-06-06:
The same for me. Sofar HYD 10ktl-3ph. Just rolled back to earlier version.

Edsol, 2025-06-06:
Same error with zcs zucchetti hp 6000 hyd Inverter.

my log:
[Home Assistant Solarman Log Jun 6 2025 (1).log](https://github.com/user-attachments/files/20624969/Home.Assistant.Solarman.Log.Jun.6.2025.1.log)

davidrapan, 2025-06-06:
Find number `0x0214` and replace it w/ `0x0485`

ssimoe50, 2025-06-06:
sorry @davidrapan, but where exactly thanks!

davidrapan, 2025-06-06:
I did edit the post.

m4t7e0, 2025-06-06:
Same Issue on my ZCS 1PH HYD 3000-6000 ZP1 (ESI 3-6K-S1)  after last extensions upgrade..
Changing  0x0214 and replace it w/ 0x0485 (in sofar_g3.yaml) on line :311  @davidrapan  solved in my case

davidrapan, 2025-06-06:
Duplicate of #647