davidrapan/ha-solarman #651: Sofar battery charge discharge

PaulOckenden, 2025-06-06:
### Description

We seem to have lost the old “ Battery Charge/Discharge Power” - there’s a new (I think) “Battery Power” but the value seems to be 1000 times too low. 

It’s probably just me, but I find it a real pain when sensors get renamed/changed. It completely messes up things like automations, average sensors, etc. Seems to happen a lot with this integration but I hardly ever see other integrations doing it. But that’s beside the point - if the new sensor worked I could start using it, but I don’t want to have to add *1000 fudge factors into my code only to have to remove them again in the future. 

### Reproduction Steps

Old sensor unavailable

### Log

Doesn’t really need a log

### Version

25.06.06

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

davidrapan, 2025-06-06:
Duplicate of #649

davidrapan, 2025-06-06:
[fix: Scaling of power entities in Sofar Wifikit](https://github.com/davidrapan/ha-solarman/commit/aa75d3ebb3f748b6f6e0fe46c7fc69186b2ffe45)

closed 2025-06-06 by davidrapan