davidrapan/ha-solarman #653: After update to v25.06.06, pylontech_force.yaml,  PylontechForce H2 v1, some sensors not available

honikos, 2025-06-06:
### Description

no sensor.pylontech_force_h2_v1_cell_0_15_temperature   up to  xxxx_0_29_temperature
and
no sensor.pylontech_force_h2_v1_cell_0_15_voltage  up to xxxx _0_29_volatage

this failure is for all battery packs from 0 to 4 (5 packs here) 

### Reproduction Steps

Update to  v25.06.06. and switch to new pylontech_force.yaml definition
Going back to old pylontech - yaml definition all is ok.

### Log

[home-assistant_solarman_2025-06-06T11-03-42.061Z.log](https://github.com/user-attachments/files/20626842/home-assistant_solarman_2025-06-06T11-03-42.061Z.log)

### Version

25.06.06

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

Pel1can111, 2025-06-06:
I just get "Failed setup, will retry: [XXXXXXXXXXXX] Device setup failed. [TypeError: unhashable type: 'list']" after updating, reverting fixes it for pylontech

davidrapan, 2025-06-06:
Please enable debug log and attach the file.

honikos, 2025-06-06:
> Hi [@honikos](https://github.com/honikos), go into device configuration and under Additional options set Modifier to 1. That means H2.
> 
> Hmm, I guess the migration doesn't work. :(

ok, it works.  All sensors are available. TOP !

davidrapan, 2025-06-07:
Hey @Pel1can111, I'm closing this one so if you still have your issue, please open then a separate ticket for it, thanks.