davidrapan/ha-solarman #661: Connecting to Pytes V5 Batteries (No Inverter)

emstone1, 2025-06-08:
### Description

I'm attempting to connect my Pytes V5 Battery to the Solarman Logger. While I see data exchange, the UI eventually displays: "Failed setup, will retry: The server has accepted the requests and is processing it, but a long duration of time will be required to do so."

I understand that Pytes V5 batteries might not be officially tested/supported with this setup, though I've seen indications of others using them. If direct connection isn't the intended use case, please feel free to close this bug report, and I'll submit a feature request.

This is a very popular battery, and I believe it's compatible with the Pylontech definition for inverters. However, I'm not seeing it work correctly when directly connected.

### Reproduction Steps

1. Connect Pytes V5 Battery directly to Solarman Logger.
2. Observe UI for connection status.
3. Note eventual error message.

### Log

`2025-06-08 16:07:30.433 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-08 16:07:50.839 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2025-06-08T22:05:14.032505+00:00', 'data': {}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JX8S73DGD7WPETFYVGE9N01P', 'minor_version': 0, 'modified_at': '2025-06-08T22:05:14.032511+00:00', 'options': {'host': '192.168.2.151', 'additional_options': {'mod': 2, 'pack': 1}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'PytesV5', 'unique_id': None, 'version': 2})
2025-06-08 16:07:50.839 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-06-08 16:07:50.841 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.2.151
2025-06-08 16:07:50.850 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [192.168.2.151, 402A8F75CFF0, 3521431156] from ('192.168.2.151', 48899)
2025-06-08 16:07:50.852 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-06-08 16:07:50.852 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Scheduling 1 query request: [{'code': 3, 'start': 0, 'end': 22, 'count': 23}] #?
2025-06-08 16:07:50.852 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Request 03 ❘ 0x03 ~ 0000 ❘ 0x0000: {'count': 23}
2025-06-08 16:07:50.853 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 b7 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 f8 15
2025-06-08 16:07:50.859 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Successful connection!
2025-06-08 16:07:54.840 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 b7 07 74 c6 e4 d1 02 01 4c fb 0e 00 6b 0e 00 00 6f 0e 37 68 05 00 d4 15
2025-06-08 16:07:54.841 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 b8 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 f9 15
2025-06-08 16:07:56.925 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 b8 08 74 c6 e4 d1 02 01 4e fb 0e 00 6d 0e 00 00 6f 0e 37 68 05 00 da 15
2025-06-08 16:07:56.925 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Request 03 ❘ 0x03 ~ 0000 ❘ 0x0000: {'count': 23}
2025-06-08 16:07:56.926 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 b9 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 fa 15
2025-06-08 16:07:58.937 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 b9 09 74 c6 e4 d1 02 01 50 fb 0e 00 6f 0e 00 00 6f 0e 37 68 05 00 e0 15
2025-06-08 16:07:58.937 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 ba 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 fb 15
2025-06-08 16:08:00.954 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 ba 0a 74 c6 e4 d1 02 01 52 fb 0e 00 71 0e 00 00 6f 0e 37 68 05 00 e6 15
2025-06-08 16:08:00.955 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-08 16:08:00.959 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-08 16:08:41.056 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2025-06-08T22:05:14.032505+00:00', 'data': {}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JX8S73DGD7WPETFYVGE9N01P', 'minor_version': 0, 'modified_at': '2025-06-08T22:05:14.032511+00:00', 'options': {'host': '192.168.2.151', 'additional_options': {'mod': 2, 'pack': 1}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'PytesV5', 'unique_id': None, 'version': 2})
2025-06-08 16:08:41.056 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-06-08 16:08:41.057 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.2.151
2025-06-08 16:08:41.066 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [192.168.2.151, 402A8F75CFF0, 3521431156] from ('192.168.2.151', 48899)
2025-06-08 16:08:41.068 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-06-08 16:08:41.068 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Scheduling 1 query request: [{'code': 3, 'start': 0, 'end': 22, 'count': 23}] #?
2025-06-08 16:08:41.068 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Request 03 ❘ 0x03 ~ 0000 ❘ 0x0000: {'count': 23}
2025-06-08 16:08:41.068 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 81 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 c2 15
2025-06-08 16:08:41.076 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Successful connection!
2025-06-08 16:08:43.108 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 81 0e 74 c6 e4 d1 02 01 7c fb 0e 00 9c 0e 00 00 6e 0e 37 68 05 00 05 15
2025-06-08 16:08:43.108 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 82 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 c3 15
2025-06-08 16:08:46.119 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 82 0f 74 c6 e4 d1 02 01 7f fb 0e 00 9f 0e 00 00 6e 0e 37 68 05 00 0d 15
2025-06-08 16:08:46.120 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Request 03 ❘ 0x03 ~ 0000 ❘ 0x0000: {'count': 23}
2025-06-08 16:08:46.120 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 83 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 c4 15
2025-06-08 16:08:49.130 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 83 10 74 c6 e4 d1 02 01 82 fb 0e 00 a2 0e 00 00 6e 0e 37 68 05 00 15 15
2025-06-08 16:08:49.131 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 84 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 c5 15
2025-06-08 16:08:52.144 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 84 11 74 c6 e4 d1 02 01 85 fb 0e 00 a5 0e 00 00 6e 0e 37 68 05 00 1d 15
2025-06-08 16:08:52.145 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-08 16:08:52.146 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
[... cut ...]
2025-06-08 16:10:13.989 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 42 1d 74 c6 e4 d1 02 01 d7 fb 0e 00 f6 0e 00 00 6e 0e 37 68 05 00 8a 15
2025-06-08 16:10:13.989 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 43 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 84 15
`

### Version

25.06.07

### Home Assistant Core Version

2025.5.3

### Home Assistant Operating System Version

15.2

emstone1, 2025-06-09:
I tried to use Pylontech which I understand is the correct profile and changed no default settings and got the same error.

Failed setup, will retry: The server has accepted the requests and it processing it, but a long duration of time will be required to do so.

`
2025-06-09 12:53:56.623 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2025-06-09T18:53:42.733526+00:00', 'data': {}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JXB0N4EDDXX2N6SKF3T840B2', 'minor_version': 0, 'modified_at': '2025-06-09T18:53:42.733532+00:00', 'options': {'host': '192.168.2.151', 'lookup_file': 'pylontech_force.yaml', 'additional_options': {'mod': 2, 'pack': 1}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'Pyte V5', 'unique_id': None, 'version': 2})
2025-06-09 12:53:56.623 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-06-09 12:53:56.625 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.2.151
2025-06-09 12:53:56.633 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [192.168.2.151, 402A8F75CFF0, 3521431156] from ('192.168.2.151', 48899)
2025-06-09 12:53:56.635 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: pylontech_force.yaml
2025-06-09 12:53:57.726 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 3, min_span: 25, max_size: 125, digits: 6, parameters: {'mod': 2, 'mppt': 4, 'l': 3, 'pack': 1}
2025-06-09 12:53:57.745 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Scheduling 5 query requests: [{'code': 3, 'start': 4096, 'end': 4106, 'count': 11}, {'code': 3, 'start': 5123, 'end': 5216, 'count': 94}, {'code': 3, 'start': 5296, 'end': 5296, 'count': 1}, {'code': 3, 'start': 5376, 'end': 5407, 'count': 32}, {'code': 3, 'start': 6144, 'end': 6175, 'count': 32}] #0
2025-06-09 12:53:57.745 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Request 03 ❘ 0x03 ~ 4096 ❘ 0x1000: {'count': 11}
2025-06-09 12:53:57.746 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 98 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 10 00 00 0b 00 cd e1 15
2025-06-09 12:53:57.757 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Successful connection!
2025-06-09 12:53:59.770 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 98 48 74 c6 e4 d1 02 01 5f 1a 10 00 d3 12 00 00 68 13 37 68 05 00 94 15
2025-06-09 12:53:59.770 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 99 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 10 00 00 0b 00 cd e2 15
2025-06-09 12:54:01.780 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 99 49 74 c6 e4 d1 02 01 61 1a 10 00 d5 12 00 00 68 13 37 68 05 00 9a 15
2025-06-09 12:54:01.780 DEBUG (MainThread) [custom_components.solarman.device] [192.168.2.151] Request 03 ❘ 0x03 ~ 4096 ❘ 0x1000: {'count': 11}
2025-06-09 12:54:01.780 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 9a 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 10 00 00 0b 00 cd e3 15
2025-06-09 12:54:03.792 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 9a 4a 74 c6 e4 d1 02 01 63 1a 10 00 d7 12 00 00 68 13 37 68 05 00 a0 15
2025-06-09 12:54:03.792 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] SENT: a5 17 00 10 45 9b 00 74 c6 e4 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 10 00 00 0b 00 cd e4 15
2025-06-09 12:54:05.806 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 9b 4b 74 c6 e4 d1 02 01 65 1a 10 00 d9 12 00 00 68 13 37 68 05 00 a6 15
2025-06-09 12:54:05.806 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-09 12:54:05.808 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Pyte V5 data in 8.063 seconds (success: False)
2025-06-09 12:54:05.808 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-09 12:54:16.023 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2025-06-09T18:53:42.733526+00:00', 'data': {}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JXB0N4EDDXX2N6SKF3T840B2', 'minor_version': 0, 'modified_at': '2025-06-09T18:53:42.733532+00:00', 'options': {'host': '192.168.2.151', 'lookup_file': 'pylontech_force.yaml', 'additional_options': {'mod': 2, 'pack': 1}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'Pyte V5', 'unique_id': None, 'version': 2})
2025-06-09 12:54:16.024 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
[... cut ...]
2025-06-09 12:54:56.802 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] RECV: a5 10 00 10 15 de 56 74 c6 e4 d1 02 01 98 1a 10 00 0c 13 00 00 67 13 37 68 05 00 5a 15
2025-06-09 12:54:56.802 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection
2025-06-09 12:54:56.803 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Pyte V5 data in 11.118 seconds (success: False)
2025-06-09 12:54:56.804 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.2.151] Closing connection`

davidrapan, 2025-06-09:
Sure, but it will require some documentation from the manufacturer.

closed 2025-07-27 by github-actions[bot]