davidrapan/ha-solarman #683: Fail on setup

Luky12568, 2025-06-22:
### Description

I am trying to set up my Sofar LSW3 but it just throws this error
[image]
My setup:
[image]

### Reproduction Steps

Just set it up

### Log

[home-assistant_solarman_2025-06-22T11-42-21.397Z.log](https://github.com/user-attachments/files/20852532/home-assistant_solarman_2025-06-22T11-42-21.397Z.log)

### Version

25.06.07

### Home Assistant Core Version

2025.6.1

### Home Assistant Operating System Version

15.2

davidrapan, 2025-06-22:
Please select appropriate profile according to [Supported Devices](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices) which is [sofar_g3](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/sofar_g3.yaml)

closed 2025-06-22 by davidrapan

Luky12568, 2025-06-22:
neither did with g3hyd

[image]

davidrapan, 2025-06-22:
Show debug logs. 😉

Luky12568, 2025-06-22:
[home-assistant_solarman_2025-06-22T17-25-23.617Z.log](https://github.com/user-attachments/files/20853864/home-assistant_solarman_2025-06-22T17-25-23.617Z.log)

Luky12568, 2025-06-22:
[image]

Luky12568, 2025-06-22:
[image]

davidrapan, 2025-06-22:
These are from switching the profile back and forth. Remove and re-add the device back again w/ correct (sofar_g3) profile.