davidrapan/ha-solarman #684: Problem with setup

MrInaccessible, 2025-06-22:
### Description

Hi,

First of all really appreciate effort that you guys put to develop this fantastic integration. To be 100% I never started it up, because I have some issues (that I will describe for a moment), but I really believe that this integration works well.

Let's back to my issue:

1. Inverter: Sofar 22KTLX-G3.
2. Wifi unit (connected via USB to the inverter) - firmware version: LSW3_15_270A_1.68
3. HA Core: 2025.6.1
4. HA Operating system: 15.2
5. Version of ha-solar integration: 25.06.07

After installation I put all required information. I tried all profiles, but of course the one that I should choose is this one: sofar_lsw3.yaml. However I've never achieved good results. It seems that integration connects to the inverter but entities aren't or wrongly reading from the unit. Example below:

[image]

Please help me!

Thanks a lot for your time and willingness of help.

Regards,
M.

### Reproduction Steps

XXX 

### Log

[home-assistant_solarman_2025-06-22T13-59-04.995Z.log](https://github.com/user-attachments/files/20853022/home-assistant_solarman_2025-06-22T13-59-04.995Z.log)

### Version

25.06.07

### Home Assistant Core Version

2025.6.1

### Home Assistant Operating System Version

15.2

davidrapan, 2025-06-22:
Hi @MrInaccessible, you need to select appropriate profile according to [Supported Devices](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices) which is [sofar_g3](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/sofar_g3.yaml)

MrInaccessible, 2025-06-22:
@davidrapan, it doesn't work with sofar_g3.yaml profile.

[home-assistant_solarman_2025-06-22T14-08-50.845Z.log](https://github.com/user-attachments/files/20853078/home-assistant_solarman_2025-06-22T14-08-50.845Z.log)

I receive message about: CRC validation failed.

As I mentioned in my first post... I tried all available profiles

davidrapan, 2025-06-22:
Then it's most likely because of #142

davidrapan, 2025-06-25:
Duplicate of #142

closed 2025-06-25 by davidrapan