davidrapan/ha-solarman #688: anenji 12kw 3 phase    sensor unknow   or data adres error

lhvandersteen, 2025-06-24:
Hello 

can sombody pleae help 
i have a anenjie 12kw 3 phase inverter hybrid,

when i download the newest version of solarman i get this error
The data address received in the request is not an allowable address for the server.

and when i download solarman 1.4 
i get sensor unknow

what can be the problem 

this is my inverter 
https://anenji.com/nl/products/12kw-48v-3-phase-off-grid-solar-inverter-parallel

davidrapan, 2025-06-25:
Hi @lhvandersteen, I'll try to add a profile for this device when I have some time. 😉

davidrapan, 2025-06-27:
Well the documentation doesn't contains any energy registers so you gonna have to create integral helpers in HA for them but file added in [feat: Add ANENJI Hybrid Inverter profile](https://github.com/davidrapan/ha-solarman/commit/aa5a2d17b7a4bf0bb3612f8fcd86a83e9b8e9d49) should provide minimal usable setup. Try it and let me know. 😉

lhvandersteen, 2025-06-28:
Hello.  Thank you.  I have received this file from them for the rs485.  Is that helpful for you 

[anenji 12kw rs485.docx](https://github.com/user-attachments/files/20959433/anenji.12kw.rs485.docx)

davidrapan, 2025-06-30:
Try remove descriptions w/ registers 223+ and see if it changes anything.

davidrapan, 2025-06-30:
Yes in `anenji_hybrid.yaml`

lhvandersteen, 2025-07-05:
i have removed the 2 descriptios 

(

      - name: "Battery"
        class: "battery"
        state_class: "measurement"
        uom: "%"
        rule: 1
        registers: [229]

       - name: "Load"
        state_class: "measurement"
        uom: "%"
        rule: 2
        registers: [225]
        icon: "mdi:percent"
)

but still the same error

[error code 1.txt](https://github.com/user-attachments/files/21076520/error.code.1.txt)

rtripon, 2025-07-14:
Hello, 
I have borrowed an ANENJI 4.2kW 24V hybrid inverter just for testing purpose from a friend.
Connected it via RS485 to an LilyGO TCAN485 and managed to extract some sensors from inverter. Currently I dont have PV input, since I got it just to win a bet :), but will hook some panels in a few days.
There is the code that's working for me, maybe it will help you or others also.
If you need some testing, I`m available, the setup was made just for testing anyway.

# https://github.com/Xinyuan-LilyGO/T-CAN485 for more details on the hardware.
#
# To use this file, include it via the Remote Packages mechanism and set the
# following in `substitutions`.
# * `rs485_uart_id`. Mandatory. Set this to match the `id` of a `uart` component
#   that you add with the same name where you configure the baud rate, parity,
#   etc., and then attach to higher-level components.
substitutions:
  name: anenji
  friendly_name: anenji
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: debug
#  baud_rate: 0
time:
  - platform: homeassistant
    id: homeassistant_time
  
api:
  encryption:

ota:
  - platform: esphome

wifi:
  ssid: ""
  password: ""
  use_address: ""

# Set pins required for LILYGO T-CAN485 board
# Source: https://github.com/Xinyuan-LilyGO/T-CAN485/issues/16#issuecomment-1872327090
# Alternative: https://gist.github.com/Bwooce/7c43622dcc9bc1fbb04d7a0f1e5528df
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

uart:
  - id: mod_bus
    baud_rate: 9600
    tx_pin: GPIO22
    rx_pin: GPIO21

modbus:
  - id: inverter_modbus
[... 988 lines cut ...]
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 100.0
    step: 0.1
    unit_of_measurement: "A"
    lambda: "return x * 0.1f;"
    write_lambda: |-
      return x * 10.0f;

  # Maximum mains charging current                                0.1A    Uint  333 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} maximum mains charging current"
    use_write_multiple: true
    address: 333
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 100.0
    step: 0.1
    unit_of_measurement: "A"
    lambda: "return x * 0.1f;"
    write_lambda: |-
      return x * 10.0f;