davidrapan/ha-solarman #704: modbus proxy connection

MarkoSchweizer, 2025-07-11:
Hi, 

I need some help to set up a modbus proxy on my Sofar HYD 3 KTL. Here is what I tried.

IP of inverter: 192.168.178.62
Port: 8899
IP of HA: 192.168.178.69

Set up the ha-modbus proxy with.

[image]

Tried to connect the integration to the modbus proxy.

[image]

But I get this error: 

The data address received in the request is not an allowable address for the server.

How do I set up the modbus proxy and the integration?

DawsonLeBrown, 2025-07-12:
Hi Marko, 

I have a HYD 5KTL-3PH - this integration works (most of the time) for me with trasport Protocol TCP - not Modbus TCP, and with profile sofar_g3hyd.yaml - not auto for your 3 KTL, maybe try profile sofar_hyd3k-6k-es.yaml?

davidrapan, 2025-08-09:
Hi @MarkoSchweizer, you need to select `sofar_g3.yaml` profile (probably). 😉