davidrapan/ha-solarman #706: Unavailable entities after Deye inverter comes back online

QaplAndroid, 2025-07-16:
Long story short:
- deye inverter goes to sleep at night
- sun comes, inverter wakes up
- home assistant integration displays values however few of them are unavailable
- reload Solarman integration - all fine

Detailed:
I have Deye 3phase low battery voltage inverter, currently working only on PV + GRID, without battery.
Every night inverter shuts off (goes to sleep mode) where datalogger is also off
As soon as sun comes, everything is back online however not everything is loaded properly, there seems to be unavailable entities.

To fix it, i need to reload integration and then all is good.

The entities that are unavailable are:
sensor.solar_deye_total_energy_import, sensor.solar_deye_total_load_consumption, sensor.solar_deye_total_production, sensor.solar_deye_total_losses, number.solar_deye_battery_equalization, number.solar_deye_battery_absorption, number.solar_deye_battery_float, number.solar_deye_battery_shutdown_voltage, number.solar_deye_battery_restart_voltage, number.solar_deye_battery_low_voltage, number.solar_deye_smartload_off_voltage, number.solar_deye_smartload_on_voltage, number.solar_deye_grid_voltage_protection_high, number.solar_deye_grid_voltage_protection_low, number.solar_deye_grid_frequency_protection_high, number.solar_deye_grid_frequency_protection_low

Versions:
Home assistant: 16.0 | 2025.7.2 
Solarman integration: 25.06.07

catcher1138, 2025-07-19:
This exact same thing happens to me also.

Home assistant version: 14.2 / 2025.7.1
Solarman: 25.06.07

Here's a paste from log:

Registrador: custom_components.solarman.parser
Fuente: custom_components/solarman/parser.py:162
integración: Solarman (documentación, problemas)
Ocurrió por primera vez: 7 de julio de 2025, 06:47:25 (8176 ocurrencias)
Última vez registrado: 10:25:24

ParameterParser.try_parse: data: {(3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 112, 7, 4927, 1, 62002, 0, 148, 677, 2, 43, 0, 0, 0, 0, 59648, 1, 19, 50776, 0, 0, 1250, 1507, 0, 0, 1, 0, 0, 0, 0, 1, 2048, 1, 13, 0, 0, 0, 0, 0, 0], (3, 586): [1250, 876, 0, 0, 0, 0, 200, 0, 0, 0, 0, 0, 2383, 2320, 2324, 0, 0, 0, 63544, 63550, 63554, 59576, 0, 5004, 841, 861, 855, 12, 10, 10, 1350, 592, 1443, 3385, 0, 0, 1350, 592, 1443, 3385, 0, 2384, 2319, 2328, 430, 440, 440, 1058, 1061, 1076, 3195, 3195, 5004, 0, 0, 0, 0, 10, 2370, 2325, 2328, 0, 0, 0, 3342, 2582, 3431, 9355, 9355, 5004, 0, 0, 0, 0, 0, 2375, 2327, 2351, 934, 929, 912, 2775, 0, 0, 0, 0, 1283, 1970, 0, 0, 4526, 28, 4165, 47, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0]}, definition: {'name': 'Total Production', 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'kWh', 'scale': 0.1, 'rule': 3, 'registers': [534, 535], 'icon': 'mdi:solar-power', 'validation': {'min': 0.1, 'dev': 100, 'invalidate_all': 2}, 'platform': 'sensor', 'key': 'total_production_sensor', 'group': 'Meter', 'update_interval': 30, 'code': 3} [ValueError("Invalidate complete dataset. total_production_sensor: 12518.400000000001 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}")]
ParameterParser.try_parse: data: {(3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 112, 7, 4927, 1, 62002, 0, 148, 677, 2, 43, 0, 0, 0, 0, 59648, 1, 19, 50776, 0, 0, 1250, 1507, 0, 0, 1, 0, 0, 0, 0, 1, 2048, 1, 13, 0, 0, 0, 0, 0, 0], (3, 586): [1250, 876, 0, 0, 0, 0, 200, 0, 0, 0, 0, 0, 2383, 2320, 2324, 0, 0, 0, 63548, 63550, 63556, 59582, 0, 5000, 838, 860, 856, 12, 10, 10, 1355, 592, 1446, 3393, 0, 0, 1355, 592, 1446, 3393, 0, 2384, 2319, 2328, 430, 440, 450, 1052, 1058, 1073, 3183, 3183, 5000, 0, 0, 0, 0, 0, 2370, 2325, 2329, 0, 0, 0, 3342, 2581, 3433, 9356, 9356, 5000, 0, 0, 0, 0, 0, 2375, 2326, 2351, 935, 931, 914, 2780, 0, 0, 0, 0, 1274, 1960, 0, 0, 4576, 28, 4215, 47, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0]}, definition: {'name': 'Total Production', 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'kWh', 'scale': 0.1, 'rule': 3, 'registers': [534, 535], 'icon': 'mdi:solar-power', 'validation': {'min': 0.1, 'dev': 100, 'invalidate_all': 2}, 'platform': 'sensor', 'key': 'total_production_sensor', 'group': 'Meter', 'update_interval': 30, 'code': 3} [ValueError("Invalidate complete dataset. total_production_sensor: 12518.400000000001 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}")]
ParameterParser.try_parse: data: {(3, 62): [6407, 4106, 6171], (3, 210): [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], (3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 117, 8, 4931, 1, 62003, 0, 161, 690, 2, 53, 0, 0, 0, 0, 59658, 1, 23, 50781, 0, 0, 1250, 1508, 0, 0, 1, 0, 0, 0, 0, 1, 2048, 1, 13, 0, 0, 0, 0, 0, 0], (3, 586): [1250, 876, 0, 0, 0, 0, 200, 0, 0, 0, 0, 0, 2371, 2331, 2324, 0, 0, 0, 63536, 63534, 63541, 59539, 0, 5000, 849, 863, 863, 12, 10, 10, 1909, 641, 1488, 4038, 0, 0, 1909, 641, 1488, 4038, 0, 2372, 2331, 2328, 430, 450, 450, 1066, 1068, 1082, 3216, 3216, 5000, 0, 0, 0, 0, 11, 2358, 2336, 2329, 0, 0, 0, 3910, 2646, 3490, 10046, 10046, 5000, 0, 0, 0, 0, 0, 2365, 2338, 2351, 935, 937, 920, 2792, 0, 0, 0, 0, 1294, 1989, 0, 0, 4576, 28, 4235, 46, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0]}, definition: {'name': 'Total Production', 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'kWh', 'scale': 0.1, 'rule': 3, 'registers': [534, 535], 'icon': 'mdi:solar-power', 'validation': {'min': 0.1, 'dev': 100, 'invalidate_all': 2}, 'platform': 'sensor', 'key': 'total_production_sensor', 'group': 'Meter', 'update_interval': 30, 'code': 3} [ValueError("Invalidate complete dataset. total_production_sensor: 12519.400000000001 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}")]
ParameterParser.try_parse: data: {(3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 117, 8, 4931, 1, 62003, 0, 161, 690, 2, 53, 0, 0, 0, 0, 59658, 1, 23, 50781, 0, 0, 1250, 1508, 0, 0, 1, 0, 0, 0, 0, 1, 2048, 1, 13, 0, 0, 0, 0, 0, 0], (3, 586): [1250, 876, 0, 0, 0, 0, 200, 0, 0, 0, 0, 0, 2366, 2334, 2325, 0, 0, 0, 63524, 63527, 63531, 59510, 0, 5004, 857, 865, 864, 12, 10, 10, 1921, 678, 1526, 4125, 0, 0, 1921, 678, 1526, 4125, 0, 2367, 2334, 2329, 440, 450, 450, 1081, 1074, 1088, 3243, 3243, 5004, 0, 0, 0, 0, 11, 2359, 2339, 2337, 0, 0, 0, 3935, 2691, 3536, 10162, 10162, 5004, 0, 0, 0, 0, 0, 2359, 2341, 2353, 933, 939, 922, 2794, 0, 0, 0, 0, 1300, 1997, 0, 0, 4676, 26, 4265, 46, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0]}, definition: {'name': 'Total Production', 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'kWh', 'scale': 0.1, 'rule': 3, 'registers': [534, 535], 'icon': 'mdi:solar-power', 'validation': {'min': 0.1, 'dev': [... cut ...]  'Total Production', 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'kWh', 'scale': 0.1, 'rule': 3, 'registers': [534, 535], 'icon': 'mdi:solar-power', 'validation': {'min': 0.1, 'dev': 100, 'invalidate_all': 2}, 'platform': 'sensor', 'key': 'total_production_sensor', 'group': 'Meter', 'update_interval': 30, 'code': 3} [ValueError("Invalidate complete dataset. total_production_sensor: 12519.400000000001 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}")]

CrazyUs3r, 2025-07-20:
hey copy the repo to your desktop and after unpacking you can copy it into custom_components

marccap2002, 2025-07-21:
Same problem

marccap2002, 2025-07-21:
For now, I've created an automation that reloads the integration every time the Deye turns on and goes from "Unavailable" to any other state.

```
alias: Reload deye solarman integration
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.solarman_device_state
    from: unavailable
    to: null
    for:
      hours: 0
      minutes: 1
      seconds: 0
conditions: []
actions:
  - action: homeassistant.reload_config_entry
    data:
      entry_id: 01JSRVG55XGXJ12JP6P76NDV0P
mode: single

```

davidrapan, 2025-09-24:
I need the debug log from the morning the inverter turns back online.

closed 2025-12-25 by davidrapan