davidrapan/ha-solarman #712: Cannot turn on generator

Ramaddan, 2025-07-23:
At some point of the upgrade cycles I lost the capability to turn on the Generator for the deye hybrid 8kw

A few months ago I could do that, but not anymore

davidrapan, 2025-08-14:
Debug logs, is it 1P or 3P, and when was the last time it worked?

Ramaddan, 2025-08-16:
Hi,

Thank you for your reply

It is 1P

It stopped a long time ago, half a year minimum

Thanks

Ramaddan, 2025-08-17:
Hi,

Ok, turned it on and downloaded it after trying many times.

But as I read through it, I can't seem to find anything related to the switch.generator trigger.

[error_log-2e.txt](https://github.com/user-attachments/files/21824227/error_log-2e.txt)

Lamer-v, 2026-09-23:
I can add a reproducible data point for this. Deye hybrid 3P, register `0x0146` bit 13 (`switch.<ip>_generator`, "Gen Force"), current integration version, profile already has the correct register (post-#843/#847 fix, not the old `0x00EA`).

**Setup:** 3 inverters connected via Solarman WiFi stick loggers, local Modbus TCP (not cloud) through `ha-solarman`.

**Test 1 - via `switch.<ip>_generator` (tap to toggle in the dashboard):**
- Entity state flips to `on` immediately, no error/traceback in the log.
- Physically standing next to the inverter at the moment of the write: **relay does not click, generator does not start.**
- Re-read the same parameter ~14 minutes later via Solarman Business (cloud, "Gen Force" field on Battery Setting-3 page): still shows `Disable` - the write never actually stuck on the device side, despite HA showing `on` with no error.

**Test 2 - raw `solarman.write_holding_register` service on the same register:**
- Throws:
  ```
  custom_components.solarman.pysolarman.umodbus.exceptions.AcknowledgeError:
  The server has accepted the requests and it processing it, but a long...
  ```
  (Modbus exception 0x05 -device-side "accepted, processing" response that the integration doesn't retry/poll to confirm completion.)

**Test 3 - control plane (baseline, works):** Same register, same bit, toggled from Solarman Business (cloud) "Battery Setting-3 → Gen Force" dropdown → Set. Relay clicks within seconds, generator starts normally (confirmed multiple times, physically observed).

So: reads are reliable, the register address is correct, and the cloud write path works consistently - but a local write from HA/`ha-solarman` for this specific register either times out with an `AcknowledgeError` (raw service call) or reports success with no visible error while silently not taking effect on the device (via the `switch` entity).

One thing that might be relevant, noted in #843 by the person who fixed the register mapping: bit 13 of `0x0146` also participates in the `Meter` select (`bit: [0, 13] → "Generator"`). If the inverter firmware expects a coherent full-register write (not just a single-bit read-modify-write) to treat it as a valid forced-start command, that could explain why cloud (which presumably writes a consistent full value) works while a naive single-bit toggle from HA doesn't reliably land. Not confirmed, just a hypothesis based on that overlap - happy to help test further if useful (I can capture debug logs around a live test).