davidrapan/ha-solarman #718: Feature Request: Extend Compatibility for Zucchetti ZCS HYD6000-ZP1 Inverter

malesci, 2025-07-29:
### Is your feature request related to a problem?

I am currently using the ha-solarman integration with a Zucchetti ZCS HYD6000-ZP1 inverter. I've noticed that when using the sofar_g3.yaml configuration (which seems to be the closest available option), not all entities related to the inverter are being detected correctly. This limits the full visibility and control of my solar PV system within Home Assistant.

### Describe the solution you'd like

I would like to request an extension of this integration's compatibility to include the Zucchetti ZCS HYD6000-ZP1 inverter model. The goal is to ensure that all relevant entities are properly exposed in Home Assistant.

### Describe alternatives you've considered

Eventually, how can I contribute?

Could you kindly provide guidance on how I can best assist? For example:

- What additional information would be helpful? (e.g., datalogger logs, specific inverter registers, technical documentation of the Solarman protocol for this model, etc.)
- Is there a preferred methodology for mapping new inverters? (e.g., reverse engineering Modbus registers, network traffic analysis, etc.)
- What is the recommended process for proposing changes or adding new configuration files?
          
I am available to provide any further details and to test new implementations.

malesci, 2025-08-17:
Hi @davidrapan,

Thanks for your reply. To answer your question, I can provide a comparison of the entities I get from two different profiles.

I am currently using **sofar_g3.yaml**. I have also tested **sofar_g3hyd.yaml**. Neither profile is complete on its own, but they seem to complement each other. It appears that a combination of the two would provide better coverage for the Zucchetti ZCS HYD6000-ZP1.

Also, I can confirm that setting the profile to **auto** results in an error and the integration fails to load.

Here is a list of the key missing entities for each profile.

**Entities Missing in sofar_g3.yaml (but available in sofar_g3hyd.yaml)**

- ActivePower_Load_R_EPS (and S, T, Total)
- ActivePower_Output_R (and S, T, Total)
- ActivePower_PCC_R (and S, T, Total)
- Current_Load_R_EPS (and S, T)
- Current_Output_R (and S, T)
- Current_PCC_R (and S, T)
- Voltage_Output_R_EPS (and S, T)
- Serial Number
- Hardware Version
- Software Version Master
- Software Version Slave
- Safety Version
- Ambient temperature 1
- Module temperature 1
- Radiator temperature 1
- Production Time Today

**Entities Missing in sofar_g3hyd.yaml (but available in sofar_g3.yaml)**

- Grid Power
- Load Power
- Power
- L1 Current
- L1 Voltage
- L2 Current
- L2 Voltage
- L3 Current
- L3 Voltage
- Passive: Grid power
- Passive: Maximum Battery power
- Passive: Minimum Battery power
- Frequency
- Load UPS Power
- Load UPS L1 Current
- Load UPS L1 Voltage

As you can see, the necessary data seems to be split between the two profiles. A new or merged profile would likely improve the integration.

I hope this breakdown is helpful for extending compatibility.

Best regards

malesci, 2025-08-17:
Hi @Guldoman,

You were absolutely right, my apologies! I've found them.

I went to "Developer Tools" -> "States" and searched for the `sensor.solarman_device_state` entity, and all the information you mentioned is there as attributes. Thank you so much for clarifying that!

This brings me back to the main issue. Do you think this profile is the correct one, or does it still need some fine-tuning to fully support this inverter?

Thanks again for your guidance!

davidrapan, 2025-08-17:
You can easily view the attributes in dashboards... and both profiles contain exactly the same entities, `sofar_g3.yaml` is just newer.

closed 2025-08-17 by davidrapan

daniporr, 2025-09-05:
hi @malesci, I have your same inverter as you and `sofar_g3` seems to work well.

However, if you find a way to improve it, please let us know so we can all benefit from your findings.

Perhaps, this discussion may be useful in this regard: https://github.com/StephanJoubert/home_assistant_solarman/discussions/148