davidrapan/ha-solarman #726: warning in the logs after update to 2025.8

Grip, 2025-08-07:
### Description

Hi,
after updating to HAOSS 2025.8.0 I receive this message in the logs:

### Reproduction Steps

just update the HAOSS to latest version

### Log

```
Logger: homeassistant.helpers.entity_platform
Source: helpers/deprecation.py:222
First occurred: 9:14:40 AM (32000 occurrences)
Last logged: 10:04:32 AM

The deprecated function platform_translations was called from solarman. It will be removed in HA Core 2026.8. Use platform_data.platform_translations instead, please create a bug report at `https://github.com/davidrapan/ha-solarman/issues```

### Version

core-2025.8.0

### Home Assistant Core Version

2025.8.0

### Home Assistant Operating System Version

Home Assistant OS 16.0

sofkaski, 2025-08-07:
@davidrapan fixed this already in 162ef44abfa93. Will be included the next `ha-solarman` release.

DimStar77, 2025-08-07:
FWIW: I have hot-patches my system with https://github.com/davidrapan/ha-solarman/commit/162ef44abfa9375d9b1119f68abaad5aceeed59c

That has immediately lowered the CPU usage back to a useful state (the CPU was clogged by logging that warning over and over)

JanChr4711, 2025-08-08:
Hello from Flensburg in Germany. I have changed the line in custom_components/solarman/entity.py, and HA has already stopped producing Gigabyte-logs. But I cannot find the file hacs.json anywhere, Should I create it - and if - where tori it?
Best regards jan