davidrapan/ha-solarman #730: CRC Validation Error When Configuring Sofar 25KTLX-G3 Inverter

trash-panda-v91-beta, 2025-08-10:
### Description

When attempting to configure my Sofar 25KTLX-G3 Inverter with a simple configuration, I encounter the following error:

```
Failed setup, will retry: CRC validation failed.
```

### Configuration Attempted
I'm using this minimal configuration which fails:

```yaml
info:
  manufacturer: Sofar
  model: G3

default:
  update_interval: 10
  digits: 6

parameters:
  - group: Device
    items:
      - name: "Room temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        rule: 2
        registers: [0x0418]
```

### Working Configuration
The same configuration works with `StephanJoubert/home_assistant_solarman` integration when I add specific Modbus request parameters:

```yaml
info:
  manufacturer: Sofar
  model: G3

default:
  update_interval: 10
  digits: 6

requests:
  - start: 0x0400
    end: 0x042E
    mb_functioncode: 0x03
  - start: 0x0480
    end: 0x04AF
    mb_functioncode: 0x03
  - start: 0x0580
    end: 0x058F
    mb_functioncode: 0x03
  - start: 0x0680
    end: 0x069B
    mb_functioncode: 0x03

parameters:
  - group: Device
    items:
      - name: "Room temperature"
        class: "temperature"
        uom: "..C"
        scale: 1
        rule: 2
        registers: [0x0418]
        icon: 'mdi:thermometer'
```

None of the predefined sofar_* definition works (I will need custom parameters anyway because I have 4 MPPT)

### Reproduction Steps

```yaml
info:
  manufacturer: Sofar
  model: G3

default:
  update_interval: 10
  digits: 6

parameters:
  - group: Device
    items:
      - name: "Room temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        rule: 2
        registers: [0x0418]
```

### Log

```log
2025-08-10 10:48:55.649 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2025-08-10T08:32:14.280964+00:00', 'data': {}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01K29HQQY82DRQANBS9SAAXVA8', 'minor_version': 0, 'modified_at': '2025-08-10T08:46:52.675397+00:00', 'options': {'host': '10.1.30.15', 'lookup_file': 'custom_parameters.yaml', 'additional_options': {'pack': 0}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'Solarman', 'unique_id': None, 'version': 2})
2025-08-10 10:48:55.649 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-08-10 10:48:55.649 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 10.1.30.15
2025-08-10 10:48:56.152 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: custom_parameters.yaml
2025-08-10 10:48:56.158 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, max_size: 125, digits: 6, parameters: {'mod': 0, 'mppt': 4, 'l': 3, 'pack': 0}
2025-08-10 10:48:56.159 DEBUG (MainThread) [custom_components.solarman.device] [10.1.30.15] Scheduling 1 query request: [{'code': 3, 'start': 1048, 'end': 1048, 'count': 1}] #0
2025-08-10 10:48:56.159 DEBUG (MainThread) [custom_components.solarman.device] [10.1.30.15] Request 03 ❘ 0x03 ~ 1048 ❘ 0x0418: {'count': 1}
2025-08-10 10:48:56.159 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] SENT: a5 17 00 10 45 40 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 18 00 01 05 3d 11 15
2025-08-10 10:48:56.163 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] Successful connection!
2025-08-10 10:48:58.646 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] RECV: a5 10 00 10 15 40 f8 1d aa 4b 89 02 01 4c 5c 98 01 ca 1a 00 00 ab 00 00 67 06 00 48 15
2025-08-10 10:48:58.646 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] SERIAL_SET: 2303437341
2025-08-10 10:48:58.646 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] SENT: a5 17 00 10 45 41 00 1d aa 4b 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 18 00 01 05 3d ad 15
2025-08-10 10:48:58.810 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] RECV: a5 13 00 10 15 41 f9 1d aa 4b 89 02 01 4f 5c 98 01 cc 1a 00 00 ab 00 00 67 01 90 02 00 03 e2 15
2025-08-10 10:48:58.810 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] SENT: a5 17 00 10 45 42 00 1d aa 4b 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 18 00 01 05 3d ae 15
2025-08-10 10:48:58.980 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] RECV: a5 13 00 10 15 42 fa 1d aa 4b 89 02 01 4f 5c 98 01 cd 1a 00 00 ab 00 00 67 01 90 02 00 03 e5 15
2025-08-10 10:48:58.980 DEBUG (MainThread) [custom_components.solarman.device] [10.1.30.15] Request 03 ❘ 0x03 ~ 1048 ❘ 0x0418: {'count': 1}
2025-08-10 10:48:58.981 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] SENT: a5 17 00 10 45 43 00 1d aa 4b 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 18 00 01 05 3d af 15
2025-08-10 10:48:59.160 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] RECV: a5 13 00 10 15 43 fb 1d aa 4b 89 02 01 4f 5c 98 01 cd 1a 00 00 ab 00 00 67 01 90 02 00 03 e7 15
2025-08-10 10:48:59.160 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] SENT: a5 17 00 10 45 44 00 1d aa 4b 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 18 00 01 05 3d b0 15
2025-08-10 10:48:59.330 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] RECV: a5 13 00 10 15 44 fc 1d aa 4b 89 02 01 50 5c 98 01 cd 1a 00 00 ab 00 00 67 01 90 02 00 03 ea 15
2025-08-10 10:48:59.330 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] Closing connection
2025-08-10 10:48:59.332 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Solarman data in 3.173 seconds (success: False)
2025-08-10 10:48:59.332 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.1.30.15] Closing connection
```

### Version

25.06.07

### Home Assistant Core Version

2025.4.4

### Home Assistant Operating System Version

-

trash-panda-v91-beta, 2025-08-10:
Here's my working [custom_parameters.yaml](https://github.com/trash-panda-v91-beta/hass-config/blob/95b73d61d8703185068be32f0daa9b4fc27e3353/custom_components/solarman/inverter_definitions/custom_parameters.yaml) I currently use with `StephanJoubert/home_assistant_solarman`

trash-panda-v91-beta, 2025-08-12:
I'm having trouble getting any entities to work with your integration. I've tried using `solar_g3.yaml` as well but no luck. 

For comparison, I'm successfully using `StephanJoubert/home_assistant_solarman` with my [custom parameters setup](https://github.com/trash-panda-v91-beta/hass-config/blob/main/custom_components/solarman/inverter_definitions/custom_parameters.yaml).

For easier debugging, I've tried with a simplified custom_parameters file that just includes "Room temperature" to make comparison between the integrations simpler.

I'd really love to switch to your integration.

sjoerdos92, 2026-04-06:
I also used a custom config for a Sofar inverter (Sofar G3 12KTLX-G3) which worked with StephanJoubert but failed with this integration. It gave the same CRC errors. I managed to fix my previous config which was based on this PR: https://github.com/StephanJoubert/home_assistant_solarman/pull/82#pullrequestreview-1072112654 by adding `requests_fine_control:` as suggested in this thread. 

After getting it to work, I made several changes. I added Export Surplus Limitation to curtail my solar panels when electricity prices are negative based on the config I found here: https://github.com/user-attachments/files/18220633/SOFAR_Modbus_Protocol_English_G3_V1.22.only.3PH.xlsx. Also, I fixed several small unit issues that were giving me warnings. I removed my old StephanJoubert solarman integration and added this one with the new config. All entity history was preserved. If anyone is interested, this is my current config in full:
```
#Sofar G3 12KTLX-G3 by Sjoerd https://github.com/StephanJoubert/home_assistant_solarman/pull/82#pullrequestreview-1072112654
#Updated April 2026 https://github.com/davidrapan/ha-solarman/issues/730
default:
  update_interval: 5
  digits: 6

requests_fine_control:
requests:
  - start: 0x0400
    end: 0x042E
    mb_functioncode: 0x03
  - start: 0x0480
    end: 0x04AF
    mb_functioncode: 0x03
#  - start: 0x0500
#    end: 0x041F
#    mb_functioncode: 0x03
  - start: 0x0580
    end: 0x0589
    end: 0x058F
    mb_functioncode: 0x03
#  - start: 0x0600
#    end: 0x060A
#    mb_functioncode: 0x03
  - start: 0x0680
    end: 0x069B
    mb_functioncode: 0x03
  - start: 0x1023
    end: 0x1024
    mb_functioncode: 0x03
parameters:
  - group: Inverter
    items:
      - name: "Inverter status"
        class: ""
        state_class: ""
        uom: ""
        scale: 1
        rule: 1
        registers: [ 0x0404 ]
        lookup:
          - key: 0
            value: "waiting"
          - key: 1
            value: "detection"
          - key: 2
            value: "grid-connected"
          - key: 3
            value: "emergency power supply"
          - key: 4
            value: "recoverable fault"
          - key: 5
            value: "permanent fault"
          - key: 6
            value: "upgrade"
          - key: 7
            value: "self-charging"
        icon: 'mdi:wrench'
      - name: "Ambient temperature 1"
        class: "temperature"
        uom: "°C"
        scale: 1
        rule: 2
        registers: [ 0x0418 ]
        icon: 'mdi:thermometer'
      - name: "Radiator temperature 1"
[... 1060 lines cut ...]
          - key: 0x0000
            value: "Disabled"
          - key: 0x0001
            value: "Enabled"
          - key: 0x0002
            value: "Balanced"
        writeback:
          register: 0x1023
          count: 2

      - name: Export Surplus Power
        platform: number
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 100
        rule: 1
        registers: [0x1024]
        range:
          min: 0
          max: 65535
        writeback:
          register: 0x1023
          count: 2
```