davidrapan/ha-solarman #741: Communication breakdown.

pkopczak, 2025-08-13:
### Description

After updating, there's no connection to my DEYE. The connection is made directly through the inverter, bypassing the DEYE data logger. Since version 10, this has stopped working. Communication is via TCP Modbus, Waveshare RS232/485 TO WIFI ETH (B)

### Reproduction Steps

Update

### Log

Rejestrator: homeassistant.helpers.entity_platform
Źródło: helpers/deprecation.py:222
Pierwsze zdarzenie: 10:41:06 (5829 wystąpień)
Ostatnio zalogowany: 10:44:04

The deprecated function platform_translations was called from solarman. It will be removed in HA Core 2026.8. Use platform_data.platform_translations instead, please create a bug report at https://github.com/davidrapan/ha-solarman/issues

### Version

25.08.12

### Home Assistant Core Version

2025.8.1

### Home Assistant Operating System Version

16

pkopczak, 2025-08-13:
Version 25.06.07 and previous ones work correctly.

davidrapan, 2025-08-13:
Hi @pkopczak, the log provided has nothing to do with your issue. Please enable debug logging and attach the file here.

MondMann2019, 2025-08-13:
Same Problem here...

pkopczak, 2025-08-13:
[home-assistant_solarman_2025-08-13T09-18-10.897Z.log](https://github.com/user-attachments/files/21750798/home-assistant_solarman_2025-08-13T09-18-10.897Z.log)

PepeOnAChair, 2025-08-13:
Exactly same problem here. "Deye SUN-12K-SG04LP3-EU" and "TCP Modbus, Waveshare RS232/485 TO WIFI ETH (B)". I also confirm that version 25.06.07 works correctly. I lost 12 hours on this bug :/

MondMann2019, 2025-08-13:
went back to version 25.08.10 and now everything ist fine again

pkopczak, 2025-08-13:
It doesn't work for me, I have to go back to 25.06.07. I have two recorders, the original one from Deye works after the update.

davidrapan, 2025-08-13:
> went back to version 25.08.10 and now everything ist fine again

Your problem was a little different as v25.08.12 requires HA Core 2025.8.0+ and I was hoping that HACS would handle it, but apparently not.

> I lost 12 hours on this bug :/

How could this happen? Didn't you check if it works correctly after the update?

> [home-assistant_solarman_2025-08-13T09-18-10.897Z.log](https://github.com/user-attachments/files/21750798/home-assistant_solarman_2025-08-13T09-18-10.897Z.log)

@pkopczak, could you please elaborate some more about your setup? So I can make more sense of it (including ips, etc.), thanks.

MondMann2019, 2025-08-13:
> Your problem was a little different as v25.08.12 requires HA Core 2025.8.0+ and I was hoping that HACS would handle it, but apparently not.
>
>
> How could this happen? Didn't you check if it works [...]

OK, thanks! I'll wait with the update until 2025.8.3...

pkopczak, 2025-08-13:
Of course. 
Inverter with original recorder.

[image]

Inverter DEYE 

[image]
Both loggers are connected to the same Inverter. Now I have a restored working version.

[image]

davidrapan, 2025-08-13:
Mhh, it looks like it's caused by [fix: Ignoring errors/exceptions during initialization](https://github.com/davidrapan/ha-solarman/commit/441c8e219a3448c8425931e4ee3eedd9003be72c)

pkopczak, 2025-08-13:
Perhaps the problem is with device identification. The original recorder has the name and serial number.
- name: "Device"
        class: "enum"
        rule: 1
        registers: [0x0000]

BrentVerhaegen, 2025-08-13:
> > Not sure if I should open a new bug report for this, but after upgrading to v2025.08.x
> 
> [v25.08.12](https://github.com/davidrapan/ha-solarman/releases/tag/v25.08.12) is for HA Core 2025.8.0+, use [v25.08.10](https://github.com/davidrapan/ha-solarman/releases/tag/v25.08.10) for previous versions of HA Core.

v25.08.10 does indeed seem to work on my HA v2025.7.3 version. 
As a suggestion, maybe you could mention in the release notes of [v25.08.12](https://github.com/davidrapan/ha-solarman/releases/tag/v25.08.12) that this requires HA Core 2025.8.0+?
I checked the release notes before upgrading, and since there weren't any breaking changes, nor a mention of HA Core 2025.8.0+ required, I upgraded the integration and assumed it would work.

Thanks for the help.

davidrapan, 2025-08-13:
It's in the `hacs.json` file and I thought HACS would handle it, but apparently not.

davidrapan, 2025-08-13:
> Nope

Yeah, I figured.. Anyway, here is [fix: Backward incompatibility of platform_data attribute](https://github.com/davidrapan/ha-solarman/commit/9dacf76ad2cd77b331b55a7c4eee6b4528ffd9f0)

davidrapan, 2025-08-13:
> Nope

Yeah, I figured.. Anyway, here is [fix: Backward incompatibility of platform_data attribute](https://github.com/davidrapan/ha-solarman/commit/cd4cbc8d7c6e970fd3b883e7ef7cfec2e2d3e82a)

fl1ck3r, 2025-08-13:
> > Nope
> 
> Yeah, I figured.. Anyway, here is [fix: Backward incompatibility of platform_data attribute](https://github.com/davidrapan/ha-solarman/commit/cd4cbc8d7c6e970fd3b883e7ef7cfec2e2d3e82a)

This worked! back online, thank you!

davidrapan, 2025-08-13:
[fix: Timeout/Error when trying to read Number of Battery packs](https://github.com/davidrapan/ha-solarman/commit/a09dca769bbabf7764715988f1d47f4e402015a8)

davidrapan, 2025-08-13:
Awesome! 🎉