davidrapan/ha-solarman #751: Power Losses sensor Sofar g3 possibly wrong

andreabafile, 2025-08-15:
### Description

Hi there, I've been recently trying to understand the power losses custom sensors that are coming with this integration, but I can't really understand the value proposed by the "Power losses" sensor for my installation which is composed of a ZCS HYD8000 and using solar.g3 profile.

However, the "Today losses" appear to be reasonable. Here below screenshot. How can I have up to 10kW of power losses? I'm wondering if all the right registers are taken into account in the "Power losses" sensor.

Thx for help.

### Reproduction Steps

[image]

[image]

### Log

n/a

### Version

25.08.16

### Home Assistant Core Version

2025.07.3

### Home Assistant Operating System Version

16.1

andreabafile, 2025-08-15:
Hi! Thank you for your help.
Here below the screenshot of the sensors that should be relevant. Maybe not the ideal moment to take as it's raining like hell, so production almost zero. If not useful, will take them again tomorrow.
Maybe, is not taking correctly into account PV_Ext and battery charge.

[image]

[image]

[image]

[image]

[image]

[image]

davidrapan, 2025-08-15:
Have you tried the newer profile (`sofar_g3.yaml`) to see if losses work in it?

andreabafile, 2025-08-17:
It looks better indeed, but I can't use that profile as some entities missing, especially for EPS. Need to stay on sofar_g3hyd.yaml

davidrapan, 2025-08-17:
They are all there, they are just "better" named.

andreabafile, 2025-08-17:
I can't see them all, I see majority but not all.
Look at EPS entities, only L1 are there. L2 and L3 missing

(N.B. they appear unavailable as I switched back to sofar_g3hyd.)

[image]

andreabafile, 2025-09-01:
Yes absolutely, indeed I see the 3 phases for Grid Load. Just missing L2 and L3 for EPS Load.

davidrapan, 2025-09-02:
I see, so I ran it again through my script and you were right, so here's the [fix: Add missing UPS/EPS entities for Sofar G3](https://github.com/davidrapan/ha-solarman/commit/70d520a11a479ede8edcdab9df43f6f5386e23a7)