davidrapan/ha-solarman #752: Support for ZCS Azzurro 1PH 6000TL-V3

senzidee, 2025-08-15:
### Is your feature request related to a problem?

Hi,
thank you for your work.

I try to switch to this module without success.
With old module I use this configuration, similar to sofar_g3.yaml -> https://gist.github.com/senzidee/28cf0bae1a1e58b5ea584a56f49dcc1e
I search in the old issue and I look up that my inverter is more similar to 3300, than to sofar_string.yaml. With sofar_string, I have no error but all values still to zero.
I try then to modify sofar_string with the registries in my old configuration but I continue to have CRC error.
This is debug log -> https://gist.github.com/senzidee/36de1bdb5fd712b58197ae65e969e2e9
And this is my configuration -> https://gist.github.com/senzidee/9d15fe9758bbbf5805942d91a860724f

### Describe the solution you'd like

I would appreciate it if the values were reported correctly.

### Describe alternatives you've considered

I didn't think of any alternatives.

davidrapan, 2025-08-15:
Hi @senzidee, you can start by removing some entity descriptions from the range that is causing the error until it works.. (registers: 0x0404 - 0x0421).

senzidee, 2025-08-16:
Hi @davidrapan, thank you for your reply. I tried to remove all registry that is causing the errors, as you suggested, but the file got empty 😅 
I tried also some things suggested in #388, I can read 404 registry with a quantity of 20, but I don't know how to interpret the data.
I have downgrade to the old integration for now but I'll be very happy to use this on the future.

davidrapan, 2025-08-16:
The range 0x0001 - 0x0027 is readable according to your log just fine. You can try adding to the profile:
```
requests_fine_control:
requests:
  - start: 0x0400
    end: 0x042E
    mb_functioncode: 0x03
  - start: 0x0480
    end: 0x04AF
    mb_functioncode: 0x03
  - start: 0x0580
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0680
    end: 0x069B
    mb_functioncode: 0x03
```