davidrapan/ha-solarman #753: New Inverter found - but inverter is already configured

bejoup, 2025-08-16:
### Description

Hey there, since the last update my Integration found a new inverter, Even if the same inverter is already configured. 
Actually don‘t know if it is a bug, but what should I do now. 
Delete the old inverter and configure the new identified with the same Name?
Greetings from Hamburg. 

### Reproduction Steps

Update via HACS, reboot, found new inverter

### Log

[image]

### Version

25.07.01

### Home Assistant Core Version

Docker

### Home Assistant Operating System Version

Docker

davidrapan, 2025-08-16:
Hi @bejoup, active discovery is a new not well tested feature and yeah it's a bug. :)

Can you please [download diagnostic data](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) and:
1. Enable debug logging
2. Restart HA
3. DIsable debug logging
4. Attach the log file here

davidrapan, 2025-08-16:
Hi @bejoup, active discovery is a new (not yet well tested) feature and yeah it's a bug. :)

Can you please:
1. [Enable debug logging](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics)
2. Restart HA
3. DIsable debug logging
4. Attach the log file here

davidrapan, 2025-08-17:
Well, the log is missing the discovery flow that triggers it, so you either deleted it or you need to wait longer after the start before ending the debug logging.

bejoup, 2025-08-17:
Guess in must have waited longer, because the second device come up a few seconds after reboot. 
I will try it again tomorrow and post the logs here. 
Is there any keyword I can search for in the log?

bejoup, 2025-08-19:
I've just realized that you can't find the inverter in the network when it's dark :(
I will try it again tomorrow. Sorry :)

bejoup, 2025-08-22:
[home-assistant_solarman_2025-08-22T14-52-51.026Z.log](https://github.com/user-attachments/files/21940200/home-assistant_solarman_2025-08-22T14-52-51.026Z.log)

davidrapan, 2025-09-03:
[fix: Discovery of already configured devices](https://github.com/davidrapan/ha-solarman/commit/f196b72701f9eee7667a6ccde9e3345120115240)

davidrapan, 2025-09-03:
[fix: Discovery of already configured devices](https://github.com/davidrapan/ha-solarman/commit/830c75e9eaacd4362109e7bc401bc9508057d155)

agarthand, 2026-09-06:
I'm seeing the same behavior here. An already existing Deye device keeps getting detected as a new device, resulting in repeated duplicates.

From the discussion, it seems that @davidrapan has already worked on a fix 830c75e. As the issue is still open and the problem persists, can someone confirm whether the fix has been included in a release?

Looking forward to seeing this resolved.

rosch100, 2026-11-01:
I found the root cause of this for my setup (HA 2026.10.4, ha-solarman 25.08.16).

**What happens**

1. The manifest uses `registered_devices: true` for DHCP, so every DHCP contact of a MAC that is already in the device registry starts a new discovery flow.
2. In `_handle_discovery` (`config_flow.py`), `async_get_device(connections={(MAC, ...)})` returns a device, but that device belongs to a different integration (here: the FRITZ!Box device tracker that tracks the inverter's WLAN MAC, same `E8FDF8` OUI as in the manifest).
3. The first branch only aborts when `device.primary_config_entry` is one of the Solarman entries. Here it is not, so the function falls through without aborting. The `else` branch (which would add the MAC to the Solarman device) never runs, because a device was found.
4. The duplicate check relies on `async_set_unique_id(DEFAULT_DISCOVERY_UNIQUE_ID)`. The configured Solarman entry was created through the user flow and has no unique ID, so `_abort_if_unique_id_configured()` does not match and a new discovery is shown again.

In my case the Solarman device also had no MAC connection at all (`connections: []`), which is why step 3 never fixed itself.

**Suggested fix**

- Only treat the device as "already configured" if it actually belongs to this integration (`any(e.domain == DOMAIN ...)` / `device.config_entries`), otherwise continue to the `else` path.
- Set the `unique_id` (or check the Solarman device identifiers) when the entry is created, so the discovery abort works for entries created via the user step.

**Workaround I applied locally:** set `unique_id` to `default_discovery_unique_id` on the configured entry, which makes `_abort_if_unique_id_configured()` block further discovery. This is a manual edit of `.storage/core.config_entries` and is not a proper fix, because a second inverter will then not be discovered automatically.