davidrapan/ha-solarman #756: Erro: Failed to setup , after HA OS upgrade

taddch, 2025-08-17:
### Description

I have today ran the latest update to HA and it broke my solarman sensors.

### Reproduction Steps

upgrade to latest HA OS. I have restored from previous backup the ran the update again and got the same result

### Log

Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 13:51:24 (3 occurrences)
Last logged: 14:02:29

Error setting up entry Inverter for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 87, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(config, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2569, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  [Previous line repeated 5 more times]
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1161, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1086, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1260, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1292, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solarman/sensor.py", line 7, in <module>
    from homeassistant.components.template.sensor import SensorTemplate
ImportError: cannot import name 'SensorTemplate' from 'homeassistant.components.template.sensor' (/usr/src/homeassistant/homeassistant/components/template/sensor.py)

### Version

20250811.0

### Home Assistant Core Version

2025.8.2

### Home Assistant Operating System Version

16.1

davidrapan, 2025-08-17:
Hi @taddch, the version you provided, nor the traceback, seems to be correct, so either you are not using the latest version or you are using a different integration.

taddch, 2025-08-17:
[image]

I have attached the screenshot showing the latest version available to download. That latest version is causing the error in the log provided. All sensors become unavailable after installing that update.

taddch, 2025-08-17:
This is what happens from when the update has run. 

[image]