davidrapan/ha-solarman #764: Load current has the wrong value.

sakrlog, 2025-08-21:
### Description

I have a Deye 8kw, using the `deye_hybrid.yaml` config. I pasted a screenshot below where the current is at 46Amps while the  load_power is 4,600 and the load_voltag is 232. And the real formula is Amps = Watts/Voltage. So, 4600.232 should be equal to ~ 20Amps

[image]

### Reproduction Steps

Integrate the deye_hybrid yaml.

### Log

I don't have debug logs. 

### Version

25.08.16

### Home Assistant Core Version

2025.8.2

### Home Assistant Operating System Version

16.1

davidrapan, 2025-08-21:
Hi @sakrlog, I'm not sure what could be wrong, but the integration is simply reading the value, so your device is probably providing the incorrect value, as you are not the only one w/ the wrong **current** readings. Did you compare the values ​​w/ the device?

sakrlog, 2025-08-21:
I did check the value on the inverter and the app of the inverter, they are correct! FYI, i have been runnin the old solarman integration before discovering you awesome project and never seen this bappen> Hi @sakrlog, I'm not sure what could be wrong, but the integration is simply reading the value, so your device is probably providing the incorrect one, as you are not the only one (e.g. #732) w/ the wrong **current** readings. Did you compare the values ​​w/ the device?

sakrlog, 2025-08-21:
So I guess you are saying that we don't alter the numbers we are getting from the inverter at all. TBH, I couldn't understand the Modifier input when registering it the first time. Could this have any effect? What is it for?

sakrlog, 2025-08-21:
Oh just noticed that the power are always the same number, although their registers are different.  

[image]
[image]

davidrapan, 2025-08-22:
I see, so is it the case only for the load or others too?

haysdb, 2025-08-22:
My inverter is also reporting the wrong values for current. It started with a firmware update a couple of months ago. Deye appears wholly unconcerned.

davidrapan, 2025-12-25:
Unfortunately, it looks like a firmware bug..