davidrapan/ha-solarman #768: Deye HV inverter can't set voltages in proper range

toruonu, 2025-08-25:
### Description

I have a Deye SUN-25K-SG01HP3 inverter, which is correctly identified:

"SG0*HP3
by Deye"

So it knows it's a HV inverter. However both the device settings as well as using the variables in developer context will not allow me to set proper voltages. Namely it has per field an upper limit of 61V while HV batteries are in the 500-700V range often. When my code tries to set relevant voltages I see log errors like this:

Error executing service: <ServiceCall number.set_value (c:01K3G0EPVTEECZAWDET95XACXK): entity_id=['number.inverter_program_1_voltage'], value=486.0>

homeassistant.exceptions.ServiceValidationError: Value 486.0 for number.inverter_program_1_voltage is outside valid range 0 - 61

It mostly has no impact as usually one drives via capacity, but as the BMS is still in calibration I can't reach the top and bottom capacities as Deye will stop charging/discharging once <1% or >99% is reached while the actual cell voltage values are far from actual edges. The best way to drive this would be via voltage until BMS calibrates properly, but I can't as range 0-61 is limiting...

Solarman in use: v25.08.16

### Reproduction Steps

Try to set voltage higher than 61V for a HV inverter on any voltage parameter (program values, max voltage etc)

### Log

[home-assistant_2025-08-25T11-55-00.287Z.log](https://github.com/user-attachments/files/21967848/home-assistant_2025-08-25T11-55-00.287Z.log)

### Version

25.08.16

### Home Assistant Core Version

2025.8.1

### Home Assistant Operating System Version

na

davidrapan, 2025-08-31:
Hi @toruonu, thank you for the report and here is a [fix: Voltage scale and range for Deye 3P](https://github.com/davidrapan/ha-solarman/commit/9303869f8ab655f29930fef5134cc7c441dcd78b)

Perfect-Web, 2025-12-23:
@toruonu no, it should be 800 and 80000, i have left you comments on the PR https://github.com/davidrapan/ha-solarman/commit/9303869f8ab655f29930fef5134cc7c441dcd78b#diff-37b369ff7adedcc93f2c0d6f6c27e873e2caa513b5c8ceb784ccdd9f24372ae9R202

[image]

But really i think the max value is 700v not 800v :)