davidrapan/ha-solarman #780: Issue with the Deye Cloud while ha-solarman is enabled

knownout, 2025-09-05:
### Description

I recently encountered a strange problem: when this integration was enabled, all batteries connected to the inverter were displayed in Deye Cloud as offline (disconnected), so the support service could not remotely update firmware. As soon as I disabled this integration, the batteries appeared in Deye Cloud as online. Could this be because the logger is being polled too often? Or what else could be the problem?

By the way, there were no disconnections or errors in HA, everything worked as expected, the only problem was with data transfer to Deye Cloud with this integration enabled.

My equipment:
1. Deye SUN-6K-SG04LP1-EU hybrid inverter
2. 3x Deye SE-G5.1 Pro-B 100A batteries
3. Solarman logger (standard logger supplied with the inverter)

### Reproduction Steps

1. Install and configure this integration
2. Wait a few hours
3. In the Deye Cloud app, all batteries connected to the inverter will be marked as disconnected

### Log

N/A

### Version

25.08.16

### Home Assistant Core Version

2025.8.3

### Home Assistant Operating System Version

16.1

marius86ar, 2025-09-05:
It's not an issue with the integration "unfortunately", it's the way the logger works. I read somewhere that the logger cannot send data to the cloud and to local device which is connected through this integration, it's limited to one client. I also don't know how to overcome this, it's frustrating.

davidrapan, 2025-09-06:
Hi @knownout, it's a known issue and is being discussed in #208

davidrapan, 2025-09-27:
Duplicate of #208

closed 2025-12-24 by davidrapan