davidrapan/ha-solarman #793: Ultima versione di Solarman

Linus-ui, 2025-09-13:
Ciao David,
ho installato l'ultima versione di Solarman,ma non mi aggiornava i valori su di un grafico,ho installato la penultima 25.08.10,il problema,sembra risolto

Linus-ui, 2025-09-13:
ti aggiorno,anche con questa versione,non ricevo informazioni aggiornate dall'inverter chint.
Se controllo i dati dell'inverter con l'applicazione chint home,vedo valori in aggiornamento dell'inverter,se li controllo con l'applicazione solarman in home assistant,non ricevo informazioni aggiornate.
Se vado nelle impostazioni della  pagina dell'integrazione ed effettuo un "ricarica",l'applicazione solarman mi restituisce i valori aggiornati,vedi foto

[image]

Linus-ui, 2025-09-13:
Buonasera David,
credo che tu ti riferisca al file che ti allego,è un autotest dell'inverter chint,altro non saprei indicarti.
Ti confermo che l'inverter funziona bene

[Autotest-2025-09-13 18-49-26.csv](https://github.com/user-attachments/files/22311916/Autotest-2025-09-13.18-49-26.csv)

davidrapan, 2025-09-13:
[Enabling debug logging](https://www.home-assistant.io/docs/configuration/troubleshooting/#enabling-debug-logging)

Linus-ui, 2025-09-14:
Buongiorno David,
spero di illustrarti quello di cui avevi bisogno,se ho capito bene,solarman utilizza una funzione che non è più usata dalla versione di HA Core 2026.8 in poi,corretto?

Registratore: homeassistant.helpers.entity_platform
Fonte: helpers/deprecation.py:222
Prima occorrenza: 05:25:51 (19703 occorrenze)
Ultimo accesso: 08:15:19

The deprecated function platform_translations was called from solarman. It will be removed in HA Core 2026.8. Use platform_data.platform_translations instead, please create a bug report at https://github.com/davidrapan/ha-solarman/issues

Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: custom_components.solarman.coordinator
Fonte: helpers/update_coordinator.py:398
Integrazione: Solarman (documentazione, problemi)
Prima occorrenza: 05:25:51 (1 occorrenza)
Ultimo accesso: 05:25:51

Timeout fetching Inverter data

Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: custom_components.solarman.parser
Fonte: custom_components/solarman/parser.py:164
Integrazione: Solarman (documentazione, problemi)
Prima occorrenza: 00:11:42 (3892 occorrenze)
Ultimo accesso: 05:25:03

ParameterParser.try_parse: data: {(3, 4097): [2350, 168, 0, 1258, 5002, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 42, 3, 0, 0, 0, 0, 3915, 0, 6948, 0, 0, 0, 736, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8769, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 107, 0, 0], (3, 4864): [0, 0, 0, 0, 0, 0, 0, 28146, 2, 30265, 0, 1250, 0, 0, 0, 0, 3, 20189, 0, 0, 0, 0, 0, 0, 0, 0, 2354, 0, 0, 0, 0, 0, 0, 0, 0, 2354, 0, 0, 0, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 79, 0, 17, 0, 180, 5002, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2354, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52, 0, 9290], (3, 6656): [17232, 21280, 21315, 17717, 19284, 19501, 12594, 12288, 0, 0, 0, 0, 0, 0, 0, 0, 12592, 12600, 14643, 12850, 13363, 12339, 14152, 0, 1049, 12342, 12337, 12337, 12338, 12849, 14647, 12341, 12338, 12333, 12592, 12852, 12592, 13104, 12338, 12338, 12336, 12341, 12338, 12589, 12337, 12852, 12339, 12849, 5500, 3600, 1000, 700, 900, 5400, 700, 5400, 2300, 4800, 990, 2, 1000, 9000, 1300, 1300, 0, 4500, 4500, 0, 2300, 5000, 5000, 5500, 1, 2730, 2730, 800, 988, 980, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 29, 3, 0, 0, 0, 0, 0, 12338, 12848, 12853, 12341, 12338, 13357, 12592, 12852, 12593, 12343, 0, 0, 0, 0, 0, 12338, 12340, 12336, 12341, 12338, 13613, 12338, 12852, 12345, 12340], (3, 8192): [63, 0, 192, 4096, 0, 0, 524, 0, 238, 0, 1250, 0, 87, 2, 4711, 0, 194, 1, 38516, 0]}, definition: {'name': 'Today Production', 'alt': 'Daily Production', 'friendly_name': "Today's Production", 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'Wh', 'scale': 1, 'rule': 3, 'registers': [4136, 4135], 'icon': 'mdi:solar-power', 'validation': {'dev': 100, 'invalidate_all': None}, 'platform': 'sensor', 'key': 'today_production_sensor', 'group': 'meter', 'code': 3} [Invalidate complete dataset - today_production_sensor validation failed, triggered by state: 736 (17736) with conditions: {'dev': 100, 'invalidate_all': None}]
ParameterParser.try_parse: data: {(3, 4097): [2349, 166, 0, 1205, 5002, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 42, 3, 0, 0, 0, 0, 3915, 0, 6948, 0, 0, 0, 736, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8769, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 107, 0, 0], (3, 4864): [0, 0, 0, 0, 0, 0, 0, 28146, 2, 30265, 0, 1200, 0, 0, 0, 0, 3, 20189, 0, 0, 0, 0, 0, 0, 0, 0, 2350, 0, 0, 0, 0, 0, 0, 0, 0, 2350, 0, 0, 0, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 79, 0, 17, 0, 180, 5002, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2353, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52, 0, 9290], (3, 6656): [17232, 21280, 21315, 17717, 19284, 19501, 12594, 12288, 0, 0, 0, 0, 0, 0, 0, 0, 12592, 12600, 14643, 12850, 13363, 12339, 14152, 0, 1049, 12342, 12337, 12337, 12338, 12849, 14647, 12341, 12338, 12333, 12592, 12852, 12592, 13104, 12338, 12338, 12336, 12341, 12338, 12589, 12337, 12852, 12339, 12849, 5500, 3600, 1000, 700, 900, 5400, 700, 5400, 2300, 4800, 990, 2, 1000, 9000, 1300, 1300, 0, 4500, 4500, 0, 2300, 5000, 5000, 5500, 1, 2730, 2730, 800, 988, 980, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 29, 3, 0, 0, 0, 0, 0, 12338, 12848, 12853, 12341, 12338, 13357, 12592, 12852, 12593, 12343, 0, 0, 0, 0, 0, 12338, 12340, 12336, 12341, 12338, 13613, 12338, 12852, 12345, 12340], (3, 8192): [63, 0, 192, 0, 0, 0, 524, 0, 229, 0, 1200, 0, 87, 2, 4711, 0, 194, 1, 38516, 0]}, definition: {'name': 'Today Production', 'alt': 'Daily Production', 'friendly_name': "Today's Production", 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'Wh', 'scale': 1, 'rule': 3, 'registers': [4136, 4135], 'icon': 'mdi:solar-power', 'validation': {'dev': 100, 'invalidate_all': None}, 'platform': 'sensor', 'key': 'today_production_sensor', 'group': 'meter', 'code': 3} [Invalidate complete dataset - today_production_sensor validation failed, triggered by state: 736 (17736) with conditions: {'dev': 100, 'invalidate_all': None}]
ParameterParser.try_parse: data: {(3, 4097): [2348, 166, 0, 1208, 5002, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 5002, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 42, 3, 0, 0, 0, 0, 3915, 0, 6948, 0, 0, 0, 736, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8769, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 107, 0, 0], (3, 4864): [0, 0, 0, 0, 0, 0, 0, 28146 [... cut ...] 'Daily Production', 'friendly_name': "Today's Production", 'class': 'energy', 'state_class': 'total_increasing', 'uom': 'Wh', 'scale': 1, 'rule': 3, 'registers': [4136, 4135], 'icon': 'mdi:solar-power', 'validation': {'dev': 100, 'invalidate_all': None}, 'platform': 'sensor', 'key': 'today_production_sensor', 'group': 'meter', 'code': 3} [Invalidate complete dataset - today_production_sensor validation failed, triggered by state: 736 (17736) with conditions: {'dev': 100, 'invalidate_all': None}]

Linus-ui, 2025-09-20:
Ciao David,
sono quelle che ti ho inviato le informazioni che ti occorrono?

Linus-ui, 2025-09-25:
Ciao David,non ho capito se ti ho fornito le informazioni che cerchi.

Linus-ui, 2025-09-26:
Non so fare quello che chiedi.
Se ricarico il dispositivo,si aggiorna correttamente,ma tutte le volte devo sempre eseguire una ricarica del dispositivo,altrimenti,non fornisce dati aggiornati

[image]

flaviosa76, 2026-01-12:
Hi David, 
i tried your integration with a chint cps inverter and I have the same issue. After some hours of work, I get the error "today_production_sensor validation failed"  and integrations blocked. Following your suggestion I removed todays_production_sensor and I can read the other values for a long time continuosly (more than one day).  Do you have any sugestion or some test that I can do to get working the production sensor too. Thank you in advance 
flavio

closed 2026-10-29 by github-actions[bot]