davidrapan/ha-solarman #799: Autodiscovery detects inverter multiple times

b-reich, 2025-09-19:
### Description

Hi, the audodiscovery finds devices which are already added to homeassistant.

### Reproduction Steps

1.  Install and set up ha-solarman.
2. Add the device via the integration UI. Set devicename to bkw and using an internal dns name for setup.
3. Confirm the device and entities are created and updating.
4. After restarting HASS its shows a new device could be added

The inverterter is located in a seperated vlan/subnet which is discoverable with an mdns proxy.

### Log

Not sure how to provides logs for auto discovery.

### Version

25.08.16 

### Home Assistant Core Version

2025.9.3

### Home Assistant Operating System Version

16.2

b-reich, 2025-09-19:
[image]

[image]

davidrapan, 2025-09-21:
Duplicate of #753

closed 2025-09-22 by b-reich