davidrapan/ha-solarman #801: deye_sting reprint 4 trillion Watt Load

kingp0dd, 2025-09-22:
### Description

This started to happen after we activated net metering

The load power values started jumping to 4T watts
Attached are screenshots
[image]
[image]
[image]
[image]

### Reproduction Steps

Install v25.08.16 
Jse deye string yml configuration
Wait until there consumption

### Log

Inverter is currently off, will upload once it gets back up

### Version

25.08.16

### Home Assistant Core Version

2025.09.3

### Home Assistant Operating System Version

16

kingp0dd, 2025-09-25:
> Hi [@kingp0dd](https://github.com/kingp0dd), try changing `rule: 1` to `rule: 2` at
> 
> [ha-solarman/custom_components/solarman/inverter_definitions/deye_string.yaml](https://github.com/davidrapan/ha-solarman/blob/7e4a59b9f87d5c395821e77737192fafc6551a05/custom_components/solarman/inverter_definitions/deye_string.yaml#L741)
> 
> Line 741 in [7e4a59b](/davidrapan/ha-solarman/commit/7e4a59b9f87d5c395821e77737192fafc6551a05)
>  rule: 1

I changed it and reloaded the integration. I'll let it run and report back. Thank you!

kingp0dd, 2025-09-27:
@davidrapan  it's been a couple of days and i haven't seen the 4T W value yet. Seems to work now!

infertux, 2026-10-05:
> I changed it and reloaded the integration. I'll let it run and report back. Thank you!

Same problem here, I just changed `rule: 1` to `rule: 2`. I'll report back whether it fixes it...

infertux, 2026-10-11:
[image]

That definitely fixed it 😃